JuliaRobotics/IncrementalInference.jl #124: autoinit vs setValKDE

mc2922, 2018-08-13:
This initialization procedure with setValKDE appears to take into account tight GPS priors correctly, whereas doautoinit appears to initialize around (0,0) only. 

```julia
for sm in [Symbol("x$i") for i in 1:5]
  IIF.doautoinit!(fg,sm)
  setValKDE!(getVert(fg,sm),kde!(getSample(getfnctype(getVert(fg, Symbol("$(sm)f1"),nt=:fnc)),N)[1]))
end
```

See: PropFac/examples/testBackwards.jl

dehann, 2018-08-24:
Thanks, this is useful information -- I will get this fixed!

pvazteixeira, 2018-09-28:
```julia
function manual_init(fgl::FactorGraph, sym::Symbol, usefct::Symbol)
  # developed when autoinit was not working properly -- should be used with care
  warn("manual_init being used as a workaround for temporary autoinit issues.")
  pts = IIF.approxConv(fgl, usefct, sym)
  X2pre = KDE.kde!(pts)
  IIF.setValKDE!(fgl, sym, X2pre)
  getData(fgl, sym).initialized = true
  nothing
end

```

dehann, 2019-03-26:
also see, eg. `manualinit!(fg, :l0, [:l0f1])`, updated in #238

dehann, 2019-05-13:
partially related to tree based initialization.

dehann, 2019-05-29:
mostly resolved through tree initialization introduced during v0.6.x

closed 2019-05-29 by dehann