JuliaRobotics/IncrementalInference.jl #126: different result from doautoinit and autoinit=true

mc2922, 2018-08-14:
autoinit keyword not working as intended?

Following code 

```julia
addNode!(fg, sym, Point2)
pp = PriorPoint2(MvNormal(pos, cov) )
addFactor!(fg, [sym;], pp, autoinit=true)
```
will throws error: BoundsError: attempt to access 0×100 Array{Float64,2} at index [1, 1]
when inferOverTree! tries to initialize first point2 :x1 again (even though it should be initialized?)

Adding one line:
```julia
addNode!(fg, sym, Point2)
pp = PriorPoint2(MvNormal(pos, cov) )
addFactor!(fg, [sym;], pp, autoinit=true)
IncrementalInference.doautoinit!(fg,sym)
```

removes the error and initializes all the point2 as expected

dehann, 2018-08-22:
Hi @mc2922 , what's supposed to happen in your first example is that `:x1` should not be initialized during the `addFactor!` call but rather wait until a second un-initialized node depends on this `:x1`.  That said, when you call `inferOverTree`, it should as a first step call `endusreAllInitialized!(fg)` which should automatically detect and do the last line you added , `doautoinit!(fg, :x1)` for you.  I will have a closer look and get back to you.  In the meantime, which version of IIF and RoME are you on?  Thanks!

mc2922, 2018-08-22:
RoME -> v0.1.6
IIF -> v0.3.6+

dehann, 2019-05-13:
Should be fixed by tree based initialization in v0.6.x release series.

dehann, 2019-05-29:
keeping open until more testing and community certainty is obtained.

dehann, 2019-06-21:
previously `batchSolve!()` was modified to `ensureAllInitialized!()` but this had other problems with n-ary factors not able to initialize properly on the previous factor graph method.  This problem should be solved by using the new tree based initialization, since upward solving is also used to initialize the variables in the factor graph.

closed 2019-06-21 by dehann