JuliaRobotics/IncrementalInference.jl #148: Option to suppress all printouts

dehann, 2018-09-04:
In case user wants to see only their output -- won't be the default.

cc @mc2922

pvazteixeira, 2018-09-10:
#160 changes most of the `print` and `println` to `info`, which can be suppressed with `logging(DevNull; kind=:info)`. 

Unfortunately, the base logging facility is a bit coarse (supports `info`, `warn`, and `error`), so it would make sense to use something like [MicroLogging.jl](https://github.com/c42f/MicroLogging.jl)

closed 2018-09-15 by dehann

dehann, 2019-06-16:
Also see new developments with CSM:
- https://github.com/JuliaRobotics/IncrementalInference.jl/blob/fcb17a53a885b62ab03a1da1d79a4147bdb97527/src/CliqStateMachine.jl#L18
- https://github.com/JuliaRobotics/IncrementalInference.jl/blob/fcb17a53a885b62ab03a1da1d79a4147bdb97527/src/SolveTree01.jl#L1592