JuliaRobotics/IncrementalInference.jl #162: Weird results on bearing range factor

pvazteixeira, 2018-09-15:
Adding a range+bearing factor on `:x21` results in a strange solution, in which `:x25` moves away from its true value and there is a significant increase in uncertainty.

`drawPoseLandms`:
[image]
[image]
[image]

Tree:
[image]

The behavior is reproducible.

dehann, 2018-11-03:
might be related to #57

dehann, 2018-11-03:
repeat note:  The issue is reproducible using this script:
https://github.com/JuliaRobotics/Caesar.jl/blob/41323d5defa03418b7b68557de0cbbbb132c339d/examples/marine/auv/Sandshark/Funky21Problem.jl#L179

dehann, 2018-11-06:
move to v0.4.5 to rapidly resolve NLsolve v3.0.0 upstream breaking changes.

dehann, 2018-12-11:
NLsolve v3.0.0 restored with #196

dehann, 2019-03-27:
This test behavior might change in versions beyond v0.5.0, especially by v0.5.5, owing to other changes made.  Important to still test against old code to understand where this error came from.

dehann, 2019-07-06:
xref #244

dehann, 2019-07-22:
Work is progressing in new CSM tree init code.  Targeting v0.7.4 or v0.7.5 to get all the required upstream and downstream changes made.

closed 2019-09-15 by dehann