JuliaRobotics/IncrementalInference.jl #165: Allow for more flexible node names

pvazteixeira, 2018-09-28:
Some node labels are longer than a single letter and number.  For example maybe one can add node labels:
```julia
:x1
:s1
:p2
:offset3
:cal10_3
:pt100_1
```

pvazteixeira, 2018-10-22:
just for reference: the same assumption is made "downstream" in [RoME/src/RobotUtils.jl:574](https://github.com/JuliaRobotics/RoME.jl/blob/10715715c7a03c5a2c1615317fe87f9b4f1da582/src/RobotUtils.jl#L574)

dehann, 2019-01-31:
cc @Affie

dehann, 2019-03-04:
xref GearsAD/GraffSDK.jl#98

dehann, 2019-03-26:
recently used `:t1` and the code worked.  The utility functions like `ls(fg)` will not work yet, but progress is being made.

dehann, 2019-05-07:
will be provided by DFG, see here:
https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/fcd5247a73b5432543afb63b9d567cb03edfc31f/src/services/GraphsDFG.jl#L231

dehann, 2019-06-12:
done

closed 2019-06-12 by dehann