JuliaRobotics/IncrementalInference.jl #237: Standardize null hypothesis in addFactor!

dehann, 2019-03-22:
for example `addFactor!(fg, ..., nullhypo=0.1)`.  Drop older `FactorNH` factor types.

keevindoherty, 2019-04-04:
I was just going to make an issue asking what the API was for this.

Is this simple enough that I could try adding it and PR? Would need to know what is required though..

dehann, 2019-04-05:
Hi Kevin, I used to use specialized `<:FunctorPairwiseNH`, but want to change to use standard `.nullhypothesis` field ([similar to newer `partials` or `multihypo` instead](https://github.com/JuliaRobotics/IncrementalInference.jl/blob/05132d174ce0692cb8952ab5bb1c343ce10eeb02/src/FactorGraphTypes.jl#L256)).  So a little wiring needed to bring in older but relevant code here:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/05132d174ce0692cb8952ab5bb1c343ce10eeb02/src/ApproxConv.jl#L179

The nullhypothesis in current mm-isam is more of a work-around than an algebraic solution.  This code is adding a large random number (i.e. add lots of entropy) to the current value for some of the particles used to approximate the marginal beliefs.  Also previously assumed to be on an Euclidean manifold.  

The hard part with null-hypothesis (at least as far as my thinking has gone) is that the Bayes (Junction) tree might need to change if factors are being removed for some random fraction of the particles.  I haven't spent time to try find a clear solution beyond the workaround.  I should add that the "just add large entropy workaround" works pretty well as a local method.

Please feel free to chime in with ideas or code fragments, even PRs if you are in the mood for getting to know the code.

Also see:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/05132d174ce0692cb8952ab5bb1c343ce10eeb02/src/ApproxConv.jl#L149

dehann, 2019-06-10:
also see #248

dehann, 2020-07-19:
Needs JuliaRobotics/DistributedFactorGraphs.jl#536

PR #791 (2020-07-25): standardize nullhypo
fixes #237

requires JuliaRobotics/DistributedFactorGraphs.jl#536
closed 2020-07-25 by dehann