JuliaRobotics/IncrementalInference.jl #261: tag v0.5.5

dehann, 2019-04-15:
@JuliaRegistrator register()

JuliaRegistrator, 2019-04-15:
Error while trying to register: Project file should contain name, uuid and version

dehann, 2019-04-15:
@JuliaRegistrator register()

JuliaRegistrator, 2019-04-15:
Registration pull request created: [JuliaRegistries/Test/41](https://github.com/JuliaRegistries/Test/pull/41)

After the above pull request is merged, it is recommended that you create
a tag on this repository for the registered package version:
```
git tag -a v0.5.5 -m "<description of version>" 6c87e77a76ba249368e0fcf8d23644ac55541894
git push v0.5.5
```

JuliaRegistrator, 2019-04-16:
Registration pull request created: [JuliaRegistries/General/49](https://github.com/JuliaRegistries/General/pull/49)

After the above pull request is merged, it is recommended that you create
a tag on this repository for the registered package version:
```
git tag -a v0.5.5 -m "<description of version>" 6c87e77a76ba249368e0fcf8d23644ac55541894
git push v0.5.5
```

dehann, 2019-04-19:
following: https://discourse.julialang.org/t/trouble-registering-package/22896/9

JuliaRegistrator, 2019-04-19:
Error while trying to register: Version 0.5.5 already exists

closed 2019-04-30 by dehann