JuliaRobotics/IncrementalInference.jl #271: [FEATURE] ignore factor until multiple factors

dehann, 2019-05-03:
When adding landmarks to a pose chain, consider a feature where the `addVariable!` call keeps the variable "hidden" until a second factor is added:
- [ ] (or some minimum user defined).  
- [ ] Feature would be similar to how marginalized variables are treated.  
- [ ] A further enhancement would be to populate the numerics of the variable at low priority until the variable appears as part of normal inference -- i.e. minimum number of factors met.

dehann, 2019-05-07:
An envisioned ability of the solver to "ignore" a variable until some criteria has been met. Most prominent example is adding single range measurements to a new landmark variable does not improve the overall SLAM solution. Therefore the solver should only update the new variable, say :l1, without "backsolving" from :l1 in an attempt to update all other system variables from measurements to :l1. The criteria could be to only fully "backsolve" from a smart variable when at least 3 factors have been attached to it.

The forward solving (proposing a value on a smart variable) can still occur at low priority.

This feature would allow the user to add and forget about "weakly connected" variables, without having to do the book keeping in front-end.

dehann, 2019-05-29:
Relevant to discussions with @tonioteran as well as mention for future notifications.

dehann, 2019-06-11:
cc @pvazteixeira

pvazteixeira, 2019-06-11:
This would be awesome! I am running into this exact situation right now where I have multiple range measurements connected to the sensor pose that may also depending on certain criteria. These only need to be projected once a solution is available. The ability to flag this when building the factor graph would be great and definitely simplify the front-end.

dehann, 2019-06-11:
All right then -- sounds like consensus to me!  I'll get on it as soon I can (IIF v0.7.0 imminant).  May ask ya'll to help test once the update is in please.

dehann, 2019-06-11:
Maybe use the name **smart variables** to designate this?  Almost the exact inverse of an "out-marginalized" variable ... another option might be the:  "in-conceivable" variables.