JuliaRobotics/IncrementalInference.jl #300: Tag v0.6.0

dehann, 2019-05-29:
@JuliaRegistrator register()

dehann, 2019-05-29:
@JuliaRegistrator register()

dehann, 2019-05-29:
@JuliaRegistrator register()

Release notes:

This is a pre-release to introduce the new clique state machine based upward solver.  Series v0.6.x introduces new features like tree based initialization as well as refactoring for the upcoming DistributedFactorGraphs.jl merger in v0.7.x.  See [the milestone](https://github.com/JuliaRobotics/IncrementalInference.jl/milestone/23?closed=1) for more info.

JuliaRegistrator, 2019-05-29:
Registration pull request created: [JuliaRegistries/General/1055](https://github.com/JuliaRegistries/General/pull/1055)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if [Julia TagBot](https://github.com/apps/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.6.0 -m "<description of version>" 97b1ba88dfccf64bafb2bbe929b51b5647053d82
git push origin v0.6.0
```

closed 2019-05-29 by dehann