JuliaRobotics/IncrementalInference.jl #350: Could we swap out fncargvID for outneighbors?

GearsAD, 2019-07-14:


dehann, 2019-07-14:
as long as the variable order for factors is maintained, yes.

Trivial example:
```julia
addFactor!(fg, [:x1;:l1], ..)

DFG.getneighbors(fg, :x1l1f1)
>> [:x1; :l1]
```

dehann, 2019-11-19:
also see new function `getVariableOrder`, must still delete `fncargvID` field.

GearsAD, 2019-11-20:
Will check, thanks. If IIF is using that then will remove `fncargvID`, but want to make sure nothing will break - probably a DFG 0.6 change.

Affie, 2020-02-11:
Should we start deprecation of the field in DFG 0.6 for `getVariableOrder`?

I wouldn't use `outneighbors`. `getVariableOrder` uses `_variableOrderSymbols`

dehann, 2020-02-20:
HI @GearsAD , just want to circle back and check if you okay with the decision to use getVariableOrder and not use outneighbors?  I don't yet have any strong feelings other than the more consistent functions are, but ditch if too much effort and the functionality already exists.

Affie, 2020-02-20:
I just want to be clear:

> Could we swap out fncargvID for outneighbors?

**No**, order is not guaranteed. 

`fncargvID` and `_variableOrderSymbols` store the same information. So that's why the deprecation in #288.

What is also nice is the graph can be rebuild from just a list of `Variables` and `Factors` at any level since `_variableOrderSymbols` are also in `SkeletonDFGFactor`

closed 2020-02-27 by dehann