JuliaRobotics/IncrementalInference.jl #363: tag v0.7.4

dehann, 2019-08-04:
@JuliaRegistrator register()

JuliaRegistrator, 2019-08-04:
Registration pull request created: [JuliaRegistries/General/2493](https://github.com/JuliaRegistries/General/pull/2493)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if [Julia TagBot](https://github.com/apps/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.7.4 -m "<description of version>" 2522145179facd7e07fa732323f23efd5d16f991
git push origin v0.7.4
```

dehann, 2019-08-05:
@JuliaRegistrator register()

JuliaRegistrator, 2019-08-05:
Registration pull request updated: [JuliaRegistries/General/2493](https://github.com/JuliaRegistries/General/pull/2493)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if [Julia TagBot](https://github.com/apps/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.7.4 -m "<description of version>" a9be75ca72c49b402eeef4f31cde3927f1ac08e4
git push origin v0.7.4
```

closed 2019-08-12 by dehann