JuliaRobotics/IncrementalInference.jl #364: Transition to LightGraphsDFG as default

dehann, 2019-08-04:
what is the best way to introduce LightGraphsDFG and Arena.  Guess the major decision is to move away from Graphs.jl to MetaGraphs.jl.  Graphs.jl is on JuliaArchive, so long term prospects won't go too far there.  Since Graphs.jl is working and has legacy I would not want just drop Graphs.jl outright.  Can we make some plan to move LightGraphs.jl primary and drop Graphs.jl to secondary.  So my suggestion is as follows:
- IIF v0.8.x is clean new API on GraphsDFG and we start making sure LightGraphsDFG is ready to take the load,
- IIF v0.9.x we switch to LightGraphsDFG as primary and demote GraphsDFG to secondary but keep around for several cycles, at least into IIF v1.0.x as secondary.

If it all goes well, IIF v1.1.x can probably deprecate GraphsDFG.  There will still be something to do about BayesTree which currently still depends on Graphs.jl and ExVertex.

dehann, 2019-08-04:
also see:
https://github.com/JuliaRobotics/IncrementalInference.jl/issues/365

dehann, 2019-09-10:
Not going to be LightGraphsDFG, but mayby LightDFG or SymbolDFG.  WIll remain GraphsDFG as default for a while, and then switch once confidence in all code surrounding new in memory graph is obtained.

Affie, 2019-09-10:
My opinion is to build on LightDFG as GraphsDFG replacement.  
Summery for comparison:
- LightDFG: DFG based directly on LightGraphs. All lightgraphs methods should be available. Keeps a label to internal id map. Factors and Variables are label indexed dictionaries.
- SymbolDFG: DFG with adjacency as a symbol dictionary. Not completely compatible with LightGraphs, therefore it will be harder to maintain. Internal structure is factor/variable label 
- MetaGraphsDFG: DFG based on metagraphs. Allows for easily adding additional meta data.

closed 2020-02-17 by dehann

dehann, 2020-10-18:
Must make sure this is no longer an issue:
https://github.com/JuliaLang/julia/issues/21356#issuecomment-293461161