JuliaRobotics/IncrementalInference.jl #368: Support LightGraphsDFG for solving

GearsAD, 2019-08-05:
REF: https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/pull/67

Affie, 2019-08-05:
@GearsAD it looks like it runs a lot deeper, I'll need to complete a course in IIF to fix.

got this so far to init with the same type :-)
```julia
#init an empty fg with a provided type
function initfg(fgtype::DataType;
                 sessionname="NA",
                 robotname="",
                 username="",
                 cloudgraph=nothing)::AbstractDFG

  fgtype <: AbstractDFG || @error "wrong type argument for initfg()"

  return fgtype(params=SolverParams())
end
```

in the mean time, this seems to fix it for now https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/pull/73

dehann, 2019-08-06:
Take this as lecture 1, intro into IIF

dehann, 2019-08-20:
related to #364

closed 2020-02-17 by dehann