JuliaRobotics/IncrementalInference.jl #398: Need to populate PPE estimate with a PPE estimate

GearsAD, 2019-09-26:
Replace this with a call to generate a PPE from the KDE: https://github.com/JuliaRobotics/IncrementalInference.jl/blob/feature/3Q19/addingppe/src/CliqStateMachineUtils.jl#L649-L650

dehann, 2019-09-26:
by PPE do you actually mean "modefit"?  I meant PPE as Parametric Point Estimates which includes mean, max, and modefit...

dehann, 2019-09-26:
intended to use:
https://github.com/dehann/IteratedProbabilisticClassification.jl

can move to JuliaRobotics also

GearsAD, 2019-09-27:
During discus sions we spoke about replacing MAP estimate with an EM-based estimate, I guess I called that PPE and inserted it into the estimate results dictionary as `:ppe`. We can change it to `:modefit`, whichever you think is best.

dehann, 2020-03-09:
PPE is properly populated since v0.10.0

closed 2020-03-09 by dehann