JuliaRobotics/IncrementalInference.jl #406: setVariablePosteriorEstimates! before solve or with autoinit

Affie, 2019-10-08:
Currently I use the (old?) method to plot the poses:
```julia
X = getKDE(v)
xmx = meanmax == :max ? getKDEMax(X) : getKDEMean(X) 
```
I'm updating to use the variable estimates dict. I don't know if I'm understanding this correctly, but I would like the variable estimates (#398) to be set before the solve is run.
The `getKDE` method works before solve, I'm guessing only with autoinit?
Or is there an 'tree init only and don't solve' kind of function that will populate the estimates?

dehann, 2019-10-08:
Two part answer.  
- The new Parametric Point Estimates (PPE) are probably not set during the autoinit yet (new name will be `graphinit`) but that should be done.
  - If any variable are not yet initialized with `graphinit`, you can force that with `ensureAllInitialized!(fg)`.  Only to be used with `autoinit` (aka `graphinit`)
- This will all be superseded by `treeinit` once that is working 100% (currently around 85% good).  The tree solve and tree init are so closely related that im working to combine them into one process with CSM.
  -  With the new upcoming `treeinit` as default, the function `ensureAllIinitialized!` will be obsolete.
  - the intended use case is to combine incremental / marginalization with treeinit for fast and reliable solutions.
  - `graphinit` requires a lot more hand-holding than should be necessary.  Therefore think of `graphinit` as a necessary evil at this point.
  - `graphinit` will be left as backup option during the early phases of `treeinit` adoption, and currently `graphinit` (still called `autoinit`) is in the early stages of that adoption process.

dehann, 2019-10-08:
the way to use `treeinit` at this point --- which can actually be done in combination with `graphinit` --- is use `addFactor!(..., autoinit=false)` for variables you wish to be initialized via treesolve.  `solveTree!` already does `treeinit` for any uninitialized variables.  This is skipped when all variables have already been initialized by `graphinit` or `manualinit`.

Again note that `treeinit` sometimes does the wrong thing, but does the right thing in many situations. The idea is to reduce reliance on `graphinit` as `treeinit` becomes more reliable.  PPE should be updated in all cases.  Just remember that `graphinit` states are very different from `solveTree!` values.  `treeinit` will remove that distinction.

dehann, 2019-10-08:
This is probably a good place to start:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/251b69cd20f02763861c67223bc0569f59a9c8db/src/FactorGraph01.jl#L753

dehann, 2019-10-08:
same for `manualinit`:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/251b69cd20f02763861c67223bc0569f59a9c8db/src/FactorGraph01.jl#L818

Affie, 2019-10-23:
See #421 for the graph init one.

Affie, 2020-04-07:
This was fixed by #241

closed 2020-04-07 by Affie