JuliaRobotics/IncrementalInference.jl #413: tag v0.7.9

dehann, 2019-10-16:
@JuliaRegistrator register

JuliaRegistrator, 2019-10-16:
Registration pull request created: [JuliaRegistries/General/4440](https://github.com/JuliaRegistries/General/pull/4440)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if [Julia TagBot](https://github.com/apps/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.7.9 -m "<description of version>" c65c26520fd606cab49e0f9e6b3457ccceb348e2
git push origin v0.7.9
```

dehann, 2019-10-17:
@JuliaRegistrator register

JuliaRegistrator, 2019-10-17:
Registration pull request updated: [JuliaRegistries/General/4440](https://github.com/JuliaRegistries/General/pull/4440)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if [Julia TagBot](https://github.com/apps/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.7.9 -m "<description of version>" c7cba3f4a595144c9855ad24af5e38d239ff6707
git push origin v0.7.9
```

JuliaRegistrator, 2019-10-17:
Registration pull request updated: [JuliaRegistries/General/4440](https://github.com/JuliaRegistries/General/pull/4440)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if [Julia TagBot](https://github.com/apps/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.7.9 -m "<description of version>" eaa790f06456bb5fbbe74b66a88dc0fbec192bff
git push origin v0.7.9
```

closed 2019-10-17 by dehann