JuliaRobotics/IncrementalInference.jl #423: Testing InMemoryTypes via ENVironment variable

dehann, 2019-10-23:
FROM SLACK:  How about using ENV variable to select whether to use GraphsDFG or LightDFG in IIF, which can allow testing for both?

dehann, 2019-10-23:
Personally not a fan of using ENV variables for default or special behavior (even in testing // letting the IIFTEST=true Travis thing slide because that is a circular dependency issue).  Settings should be local to the process only, and at the very least all be in the same place -- at this stage the AbstractDFG container.  We can definitely repeat tests for all the DFG platforms (and should do so in IIF soon), but do not particularly want to introduce ENV dependency there.  Motivation is that users will go look everywhere to learn how to use the code and then pick up the habit of using ENV variables to change the behavior of the code, which I would like to avoid.

I agree though that ENV variables are convenient, but I'm allergic to the PATH problems that Python has precisely because they rely on ENV variables.

Affie, 2019-10-23:
Ok. Biggest advantage is that it can run tests in parallel but not a big one. So maybe change the title to something like improve end to end testing on LightDFG (and CGDFG).

dehann, 2019-10-23:
sure, parallel would be nice which is a fair comment.  Trade-off in behavior vs testing time.  Can come back to it if things really too slow.  But I would ask to be very explicit in the testing code where ENV might show up to express this as a testing optimization only, and general practice of ENV is discouraged?  Unless people feel strongly the opposite.

dehann, 2019-10-23:
again reasoning is that ... when changing IIF parameters, it happens only in one place and not some in a file, some in the code start up sequence, and others in ENV.  For me this is a convention habit thing, but don't want to draw a hard line in "other / sensible" cases.

Affie, 2020-01-06:
I agree with you on setting parameters. The idea was just to test for both drivers as we transition and then remove when GraphsDFG is deprecated. We can change `const InMemDFGType` to LightDFG at some stage.

closed 2020-07-19 by dehann