JuliaRobotics/IncrementalInference.jl #443: Documenting CSM and inference steps for debugging

dehann, 2019-11-20:
### Console

```julia
# save clique subgraphs during inference at critical steps
getSolverParams(fg).dbg = true
# record CSM steps for later analysis
tree, smt, hist = solveTree!(fg, recordcliqs=ls(fg))

# print to screen the CSM steps for clique with frontal :x2
printCliqHistorySummary(tree, :x2)

# where is everything saved???
getLogPath(fg)

clid = getCliq(tree, :x5).index
STEP = 8
fncAS = hist[clid][STEP][3]
dataAS = deepcopy(hist[clid][STEP][4])
fncAS(dataAS)
```

- `solveTree!` will always generate content at `getLogPath(fg)` location.  
- More or less will be stored there depending `getSolverParams(fg).dbg`.
- `recordcliqs=[:x1;:x2;...]` will save the CSM history to RAM.

### Animate CSM Video

Can also make concurrent video of CSM at work:
```julia
csmAnimate(fg,tree,ls(fg), frames=1000)

run(`ffmpeg -r 10 -i /tmp/caesar/csmCompound/csm_%d.png -c:v libtheora -vf fps=25 -pix_fmt yuv420p -vf "scale=trunc(iw/2)*2:trunc(ih/2)*2" -q 10 /tmp/caesar/csmCompound/out.ogv`)

@async run(`totem /tmp/caesar/csmCompound/out.ogv`)
```

Also see `csmAnimate`'s own documentation.

dehann, 2019-11-20:
cc @tonioteran @Affie

dehann, 2019-11-23:
NOte that when using `getSolverParams(fg).dbg=true` each of the major stages of the CSM clique inference subgraphs are stored to the `getLogPath(fg)` destination -- including:
- `fg_build`, as copied from main fg object
- `fg_beforeupsolve`
- `fg_afterupsolve`
- `fg_beforedownsolve`
- `fg_afterdownsolve`

which shows the belief messages passed up or down in each case added as `MsgPriors` to the clique subgraphs.

Affie, 2020-08-20:
Is there a way to debug with a failed solve (where it errors and doesn't return)?

dehann, 2020-08-20:
Hi @Affie , yeah so the best I have been able to do is as follows:

First thing is to find where the logs are going to land:
```julia
@show getLogPath(fg)
```

Then solve with all debug options:
```julia
getSolverParams(fg).dbg = true
getSolverParams(fg).async = true

tree, smt, hist = solveTree!(fg, verbose=true, recordcliqs=ls(fg))
```
Verbose is very useful, but only prints to the console so that you can get an idea of how far each CSM gets and in which functions the error might occur.

With this you will then get process 1 context back while the solver works in the background -- do not start a second solve while the first is still working in the background.

Once the failures occur, there are three things to do:
- check the logpath for all details, the error stack will be printed out there in text files.
- check `smt` on which CSM failed, and you should be able to see the error stack there as well (but not always),
- for the CSM objects that have not failed but did complete, you can fetch the recordcliq history with `hist4 = fetch(smt[4])`.

One more thing you can do is to limit the number of iterations for all CSM:
```julia
getSolverParams(fg).limititers = 13
```
which allows you to force early termination of CSM and thereby get the csmc object nearest the failure.  It is actually possible to get even closer, one CSM at a time:
```julia
fg = generateCanonicaFG_Hexagonal()
@show getLogPath(fg)

tree = wipeBuildNewTree!(fg)
drawTree(tree, show=true)

getSolverParams(fg).dbg = true
getSolverParams(fg).async = true

getSolverParams(fg).limititers = 13
cliqtask = solveCliq!(fg, tree, :x2, verbose=true, recordcliq=true, async=true)

getSolverParams(fg).limititers = 9
cliqtask = solveCliq!(fg, tree, :x0, verbose=true, recordcliq=true, async=true)
...
```

dehann, 2020-08-20:
Actually there is one more you can do with `hist::Dict` (automatically fetched for all CSM if `async=false`), use `printCliqHistorySequential` to see output like this:
```julia
using RoME
fg = generateCanonicalFG_Hexagonal()
tree, smt, hist = solveTree!(fg, recordcliqs=ls(fg));
printCliqHistorySequential(hist)

```

semicolon is recommended :-)

```
julia> printCliqHistorySequential(hist)
1.1    10:27:13.868 testCliqCanRecycle 0 | null        P ----         C 2null 3null 4null |S| 
1.2    10:27:14.303 testCliqCanIncremt 0 | null        P ----         C 2null 3null 4null |S| 
1.3    10:27:14.31  isCliqUpSolved     0 | null        P ----         C 2null 3null 4null |S| 
1.4    10:27:14.362 buildCliqSubgraph  0 | null        P ----         C 2null 3null 4null |S| 
1.5    10:27:15.617 isCliqNull         0 | null        P ----         C 2null 3null 4null |S| 
1.6    10:27:15.908 doesCliqNeeddownms 0 | null        P ----         C 2null 3null 4null |S| 
1.7    10:27:15.917 determineCliqNeedD 0 | null        P ----         C 2null 3null 4null |S| 
1.8    10:27:15.955 slowCliqIfChildren 0 | null        P ----         C 2null 3null 4null |S| 
2.1    10:27:15.987 testCliqCanRecycle 0 | null        P 1null        C 5null 6null |S| 3null 4null 
2.2    10:27:15.991 testCliqCanIncremt 0 | null        P 1null        C 5null 6null |S| 3null 4null 
2.3    10:27:15.993 isCliqUpSolved     0 | null        P 1null        C 5null 6null |S| 3null 4null 
2.4    10:27:15.995 buildCliqSubgraph  0 | null        P 1null        C 5null 6null |S| 3null 4null 
2.5    10:27:15.999 isCliqNull         0 | null        P 1null        C 5null 6null |S| 3null 4null 
2.6    10:27:16.003 doesCliqNeeddownms 0 | null        P 1null        C 5null 6null |S| 3null 4null 
2.7    10:27:16.007 checkIfCliqNullBlo 0 | null        P 1null        C 5null 6null |S| 3null 4null 
2.8    10:27:16.086 blockCliqSiblingsP 0 | null        P 1null        C 5null 6null |S| 3null 4null 
2.9    10:27:16.151 determineCliqNeedD 0 | null        P 1null        C 5null 6null |S| 3null 4null 
2.10   10:27:16.155 slowCliqIfChildren 0 | null        P 1null        C 5null 6null |S| 3null 4null 
3.1    10:27:16.159 testCliqCanRecycle 0 | null        P 1null        C ---- |S| 2null 4null 
3.2    10:27:16.165 testCliqCanIncremt 0 | null        P 1null        C ---- |S| 2null 4null 
3.3    10:27:16.168 isCliqUpSolved     0 | null        P 1null        C ---- |S| 2null 4null 
3.4    10:27:16.17  buildCliqSubgraph  0 | null        P 1null        C ---- |S| 2null 4null 
3.5    10:27:16.337 isCliqNull         0 | null        P 1null        C ---- |S| 2null 4null 
3.6    10:27:16.342 doesCliqNeeddownms 0 | null        P 1null        C ---- |S| 2null 4null 
3.7    10:27:16.347 checkIfCliqNullBlo 0 | null        P 1null        C ---- |S| 2null 4null 
3.8    10:27:16.352 blockCliqSiblingsP 0 | null        P 1null        C ---- |S| 2null 4null 
3.9    10:27:16.358 determineCliqNeedD 0 | null        P 1null        C ---- |S| 2null 4null 
3.10   10:27:16.363 getBetterName7b    0 | null        P 1null        C ---- |S| 2null 4null 
4.1    10:27:16.39  testCliqCanRecycle 0 | null        P 1null        C ---- |S| 2null 3null 
4.2    10:27:16.392 testCliqCanIncremt 0 | null        P 1null        C ---- |S| 2null 3null 
4.3    10:27:16.394 isCliqUpSolved     0 | null        P 1null        C ---- |S| 2null 3null 
4.4    10:27:16.397 buildCliqSubgraph  0 | null        P 1null        C ---- |S| 2null 3null 
4.5    10:27:16.449 isCliqNull         0 | null        P 1null        C ---- |S| 2null 3null 
4.6    10:27:16.454 doesCliqNeeddownms 0 | null        P 1null        C ---- |S| 2null 3null 
4.7    10:27:16.459 checkIfCliqNullBlo 0 | null        P 1null        C ---- |S| 2null 3null 
4.8    10:27:16.465 blockCliqSiblingsP 0 | null        P 1null        C ---- |S| 2null 3null 
4.9    10:27:16.47  determineCliqNeedD 0 | null        P 1null        C ---- |S| 2null 3null 
4.10   10:27:16.476 getBetterName7b    0 | null        P 1null        C ---- |S| 2null 3null 
5.1    10:27:16.481 testCliqCanRecycle 0 | null        P 2null        C ---- |S| 6null 
5.2    10:27:16.483 testCliqCanIncremt 0 | null        P 2null        C ---- |S| 6null 
5.3    10:27:16.486 isCliqUpSolved     0 | null        P 2null        C ---- |S| 6null 
5.4    10:27:16.488 buildCliqSubgraph  0 | null        P 2null        C ---- |S| 6null 
5.5    10:27:16.493 isCliqNull         0 | null        P 2null        C ---- |S| 6null 
5.6    10:27:16.499 doesCliqNeeddownms 0 | null        P 2null        C ---- |S| 6null 
5.7    10:27:16.504 checkIfCliqNullBlo 0 | null        P 2null        C ---- |S| 6null 
5.8    10:27:16.51  blockCliqSiblingsP 0 | null        P 2null        C ---- |S| 6null 
5.9    10:27:16.515 determineCliqNeedD 0 | null        P 2null        C ---- |S| 6null 
5.10   10:27:16.52  getBetterName7b    0 | null        P 2null        C ---- |S| 6null 
6.1    10:27:16.525 testCliqCanRecycle 0 | null        P 2null        C ---- |S| 5null 
6.2    10:27:16.528 testCliqCanIncremt 0 | null        P 2null        C ---- |S| 5null 
6.3    10:27:16.53  isCliqUpSolved     0 | null        P 2null        C ---- |S| 5null 
6.4    10:27:16.532 buildCliqSubgraph  0 | null        P 2null        C ---- |S| 5null 
6.5    10:27:16.537 isCliqNull         0 | null        P 2null        C ---- |S| 5null 
6.6    10:27:16.543 doesCliqNeeddownms 0 | null        P 2null        C ---- |S| 5null 
[... cut ...]
6.17   10:27:49.298 doCliqDownSolve    0 | upsolved    P 2downsolved  C ---- |S| 5downsolved 
4.18   10:27:50.247 exitStateMachine   0 | downsolved  P 1downsolved  C ---- |S| 2downsolved 3downsolved 
2.24   10:27:50.251 exitStateMachine   0 | downsolved  P 1downsolved  C 5downsolved 6downsolved |S| 3downsolved 4downsolved 
5.18   10:27:50.257 exitStateMachine   0 | downsolved  P 2downsolved  C ---- |S| 6downsolved 
6.18   10:27:50.348 exitStateMachine   0 | downsolved  P 2downsolved  C ---- |S| 5downsolved 
```

dehann, 2020-08-20:
The most pressing developer tool here is to better manage how CSM terminates on error.  The two options I have in mind are either:
- Propagate the error all the way (less desirable)
- Build a dedicated error state in CSM which records and closes out a CSM on error and the notify other CSMs to stop as well.

dehann, 2020-08-20:
reminder that the revamped `getSolverParams(fg).useMsgLikelihoods = true` is default false at this time and will be made default true in future.

dehann, 2020-08-20:
Another trick that doesn't come up too often but none the less: `getSolverParams(fg).multiproc = false` -- is default true.

dehann, 2020-08-20:
Also worth stating again here, the purpose of `recordcliqs` is to allow repeat steps:
```julia
# either from hist or
hist4 = fetch(cliqtask[4])

# the next function that would have been called at step 11
fnc11 = hist4[11][3]
csmc11 = deepcopy(hist4[11][4])

# must reset the condition
getSolveCondition(csmc11.cliq) = Condition()

# now redo that step (likely with debugger)
fnc12_ = fnc11(csmc11)
```

dehann, 2020-08-29:
added a new tool that allows early stopping of any CSMs per user desire, making it much easier to debug how CSMs work:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/1ad9b34579d9bb6fbbe8ea317ba8d4ff66b241e7/test/testBasicTreeInit.jl#L8

dehann, 2020-09-10:
Snapshot example of hex init:
https://github.com/JuliaRobotics/Caesar.jl/blob/a486c0728139e485ded88fa7f725295add002a7a/examples/dev/devCascadeInitCSM.jl#L3

dehann, 2020-09-11:
Commit 426e98f adds a new swim lane type CSM printout:

```julia
julia> IIF.printCSMHistoryLogical(hists)
    | 1   x0           | 2   x4           | 3   l1           | 4   x1           | 5   x5           | 6   x3           
----+------------------+------------------+------------------+------------------+------------------+------------------
1   | 1   testCli null | 2   testCli null | 3   testCli null | 4   testCli null | 5   testCli null | 6   testCli null 
2   | 7   testCli null | 8   testCli null | 9   testCli null | 10  testCli null | 11  testCli null | 12  testCli null 
3   | 13  isCliqU null | 14  isCliqU null | 15  isCliqU null | 16  isCliqU null | 17  isCliqU null | 18  isCliqU null 
4   | 19  buildCl null | 20  buildCl null | 21  buildCl null | 22  buildCl null | 23  buildCl null | 24  buildCl null 
5   | 25  canCliq null | 26  canCliq null | 27  canCliq null | 28  canCliq null | 29  canCliq null | 30  canCliq null 
6   | 31  blockUn null | 32  blockUn null | 33  blockUn null | 34  blockUn null | 35  blockUn null | 36  blockUn null 
7   | 37  traffic null | 38  traffic null | 39  traffic null | 40  traffic null | 41  traffic null | 42  traffic null 
8   | 43  determi null | 44  checkIf null | 45  checkIf null | 46  checkIf null | 47  checkIf null | 48  checkIf null 
...
```

dehann, 2020-09-13:
as of FSM v0.2.8 a new `injectDelayBefore` as well as a watchdog `timeout` features have been added to help with the debugging process.  Note that watchdog `timeout` is not really meant for mainstream use in CSM just yet, more to help development when CSMs don't terminate by themselves.

dehann, 2020-09-18:
Just copying a recent example here for the record:
```julia
using Caesar, RoME

fg = generateCanonicalFG_Hexagonal(graphinit=false)

getSolverParams(fg).treeinit = true
getSolverParams(fg).graphinit = false
getSolverParams(fg).limititers = 100

getSolverParams(fg).drawtree = true
getSolverParams(fg).showtree = true
getSolverParams(fg).dbg = true
getSolverParams(fg).async = true

# limitcliqs = [:x0=>8;:x4=>12;:l1=>21;:x1=>21;:x5=>50;:x3=>50] # breaks
# limitcliqs = [:x0=>8;:x4=>13;:l1=>21;:x1=>21;:x5=>60;:x3=>60] # 50 # doesnt break, blocks

# injectDelayBefore=[2=>(canCliqMargRecycle_StateMachine=>5); ] # step 8
# injectDelayBefore=[5=>(canCliqMargRecycle_StateMachine=>5); ]
# injectDelayBefore=[6=>(towardUpOrDwnSolve_StateMachine=>10); ]
# injectDelayBefore = nothing

mkpath(getLogPath(fg))
verbosefid = open(joinLogPath(fg, "csmVerbose.log"),"w")
# verbosefid = stdout
tree, smt, hists = solveTree!(fg, recordcliqs=ls(fg), verbose=true, verbosefid=verbosefid, timeout=50 ) #, timeout=40 , injectDelayBefore=injectDelayBefore ) #, limititercliqs=limitcliqs);

flush(verbosefid)
close(verbosefid)
# for .async = true (because .drawTree=true)
smt[7] |> x->schedule(x, InterruptException(), error=true)

open(joinLogPath(fg, "csmLogicalReconstructMax.log"),"w") do io
  IIF.reconstructCSMHistoryLogical(getLogPath(fg), fid=io)
end

# async case
fetchCliqHistoryAll!(smt, hists)

open(joinLogPath(fg, "csmSequ.log"),"w") do fid
  printCliqHistorySequential(hists, nothing, fid)
end

open(joinLogPath(fg, "csmLogi.log"),"w") do fid
  printCSMHistoryLogical(hists, fid)
end

# printCliqHistorySequential(hists)
# printCliqHistorySequential(hists, 1=>10)
# printCliqHistorySequential(hists, [1,4,6]=>11:15)
# printCliqHistorySequential(hists, [1=>9:16; 2=>20:34; 4=>29:34])
# printCliqHistorySequential(hists, [5=>12:21;6=>12:21])

printCSMHistoryLogical(hists)

# also see dbg logs at this path for more info
# @show getLogPath(fg)

using Images
csmAnimateSideBySide(tree, hists, encode=true, nvenc=true, show=true)
```

Affie, 2020-09-28:
Enable clique history recording and provide smtasks, then retrieve with `fetchCliqHistoryAll!` even if an exception is thrown.
```julia
smtasks = Task[]
solveTree!(fg; smtasks, verbose=true, timeout=10, recordcliqs=ls(fg));
hists = fetchCliqHistoryAll!(smtasks);
```
We can automate it further in the future.

dehann, 2020-09-29:
Will the internal `fetchCliqHistoryAll!` work in general or should we definitely call it after a solve?
See it has be commented out:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/8bede364027b62eb1e3dba1a5b2285de583c6050/src/SolverAPI.jl#L533

Affie, 2020-09-29:
> Will the internal fetchCliqHistoryAll! work in general or should we definitely call it after a solve?

For now, it has to be called after the solve with `smtasks` passed in by reference. I was thinking to still returns if dbg is on.
So it would not throw an error but `@error` and return the normal `tree, smtasks, hists`

I don't know about the commented out call.

dehann, 2020-09-30:
On `Circle(8,...)` case...

```julia
julia> printCliqHistorySequential(hists, [8=>20:33; 4=>31:34])
4.31   04:05:48.385 tryInitCliq          | needdownmsg P 2initialized C 8needdownmsg |S| 6upsolved 
8.20   04:05:48.406 trafficRedirectCon   | needdownmsg P 4initialized C ---- |S| 
8.21   04:05:48.412 doAllSiblingsNeedD   | needdownmsg P 4initialized C ---- |S| 
8.22   04:05:48.417 determineCliqNeedD   | needdownmsg P 4initialized C ---- |S| 
8.23   04:05:48.422 dwnInitSiblingWait   | needdownmsg P 4initialized C ---- |S| 
8.24   04:05:48.428 attemptDownInit      | needdownmsg P 4initialized C ---- |S| 
8.25   04:05:48.474 rmMsgLikelihoodsAf   | needdownmsg P 4initialized C ---- |S| 
8.26   04:05:48.479 decideUpMsgOrInit    | needdownmsg P 4initialized C ---- |S| 
8.27   04:05:48.484 attemptCliqInitUp    | needdownmsg P 4initialized C ---- |S| 
8.28   04:05:48.489 prepInitUp           | needdownmsg P 4initialized C ---- |S| 
8.29   04:05:48.496 tryInitCliq          | needdownmsg P 4initialized C ---- |S| 
4.32   04:05:48.501 slowWhileInit        | initialized P 2initialized C 8needdownmsg |S| 6upsolved 
8.30   04:05:49.035 rmUpLikeliSaveSubF   | needdownmsg P 4initialized C ---- |S| 
8.31   04:05:49.04  checkUpsolveFinish   | needdownmsg P 4initialized C ---- |S| 
8.32   04:05:49.045 trafficRedirectCon   | needdownmsg P 4initialized C ---- |S| 
8.33   04:05:49.05  doAllSiblingsNeedD   | needdownmsg P 4initialized C ---- |S| 
4.33   04:05:49.096 prepInitUp           | initialized P 2initialized C 8needdownmsg |S| 6upsolved 
4.34   04:05:49.1   tryInitCliq          | initialized P 2initialized C 8needdownmsg |S| 6upsolved 
```

Affie, 2020-10-27:
Everything logged with `logCSM` as debug level can be displayed on the console by enabling debug printing on the clique, for example, enable CSM 2 with `ENV["JULIA_DEBUG"] = :csm_2`

Affie, 2021-02-25:
```julia
getSolverParams(fg).async = true
smtasks = Task[]
solveTree!(fg; smtasks)
```
Then if it hangs call:
```julia
IIF.throwIntExcToAllTasks(smtasks)
```

Affie, 2021-02-25:
`getSolverParams(fg).dbg=true` also saves the `csmc` and `statemachine` in task storage.
it is saved in:
`smtasks[i].storage[:csmc]` 
`smtasks[i].storage[:statemachine]`