JuliaRobotics/IncrementalInference.jl #458: Confirm clique sub-graphs only use priors once

Affie, 2019-11-27:
### The tree
[image]

### fg_built, see `l0f1`
[image]

### fg_beforeupsolve look like it is used
[image]

dehann, 2019-11-28:
yes this is a bug, the `:l0f1` should only occur once.  have to double check if it should first occurrence or frontals only.  The hard part is how do this for initialization.

Affie, 2019-12-06:
By the same logic: Should all factors not only be used once? On frontals.
see `x3lm3f1`
Note, I'm only looking at subgraphs fg_beforeupsolve (message priors included) so don't know what else the code might do.
**Tree**
[image]
**Clique 1 - x3,lm0,lm3,x1**
[image]

**Clique 2 - x2: x1,x3,lm3**
[image]

Affie, 2019-12-08:
just copying here for now:
```julia
# create subgraph with no factors on/between seperators
function buildSubgraphFromLabels!(dfg::AbstractDFG,
                                  cliqSubFg::AbstractDFG,
                                  frontals::Vector{Symbol},
                                  separators::Vector{Symbol};
                                  solvable::Int=0)

  for sym in separators
    DFG.addVariable!(cliqSubFg, deepcopy(DFG.getVariable(dfg, sym)))
  end

  addfac = Symbol[]
  for sym in frontals
    DFG.addVariable!(cliqSubFg, deepcopy(DFG.getVariable(dfg, sym)))
    append!(addfac, getNeighbors(dfg,sym))
  end

  allvars = ls(cliqSubFg)
  for sym in addfac
    fac = DFG.getFactor(dfg, sym)
    vos = fac._variableOrderSymbols
    if !exists(cliqSubFg,fac) && vos ⊆ allvars   #TODO don't add duplicates to start with
      DFG.addFactor!(cliqSubFg, fac._variableOrderSymbols, fac)
    end
  end

  # remove orphans if any are there by mistake
  for fct in DFG.getFactors(cliqSubFg)
    # delete any neighboring factors first
    if length(getNeighbors(cliqSubFg, fct)) != length(fct._variableOrderSymbols)
      DFG.deleteFactor!(cliqSubFg, fc)
      @error "deleteFactor! this should not happen"
    end
  end

  return cliqSubFg
end
```

dehann, 2019-12-19:
Hi @Affie , thanks two things:
- trying to consolidate the now three different buildsubgraph functions is been bit of a lift, and we still have the three separate functions, although they are together in DFG v0.5.2 now.  For future, work I'd request to please help look at how we can drop the 3 build sub graph functions down to just one common function.
- The examples in this issue is great thanks, although I do not have access to the factor graphs you used to make them.  Could you perhaps indicate what the original factor graphs look like please?

dehann, 2019-12-19:
attempt to recreate the first:
```julia
using IncrementalInference

fg = initfg()

addVariable!(fg, :x0, ContinuousScalar)
```

dehann, 2019-12-19:
As suggested by @Affie , factors on separators should not be added.

dehann, 2020-01-05:
Ensuring that we properly resolve Ex2

PR #472 (2019-12-19): sub-graphs with only factors on frontals connected
This may be a partial fix for the test and fixes #458 (possibly)

Do not merge without looking at the effect of not having factors in cliques
commit 96a2263d (2019-12-25): fix #458
closed 2020-01-05 by dehann