JuliaRobotics/IncrementalInference.jl #459: Consolidate message priors

Affie, 2019-11-28:
It looks like parametric and mm can fit in a standard message type. Since the standard variable node data is used for now.
Can we have a look at what might be needed in the future and define a message type. I see:
 - EasyMessage
 - TempBeliefMsg - used
 - NBPMessage

I'll base parametric on it then. We can use it a bit and then update mm when we are happy.

dehann, 2019-12-03:
Hi, yes definitely need to consolidate both the belief messages as well as the channels over which the messages are sent in the Bayes tree CSM structures.

dehann, 2019-12-03:
see edit comments in first post.

Affie, 2020-01-15:
Copied from https://github.com/JuliaRobotics/IncrementalInference.jl/pull/532#issuecomment-574268523 to have related comments in one issue: 

> Hi, i don’t completely understand. How do you want to solve a clique in isolation? Do you manually send the messages?

That was a bit unclear sorry -- I'm thinking of the `recordcliqs` interface of CSM/FSM which records state machine steps in the `hist::Vector` container.  The dream here is to be able to store/restore each individual step of CSM with enough context to run in isolation.  For example, think about optimizing or refactoring CSM: if we can store one transition from a working sample, then we can debug and change individual steps simultaneously across all applications.  

My thinking around messages in cliques or edges is inspired by this "clique in isolation" or "individual CSM steps" way of thinking.  The knee jerk reaction is that a clique is a self contained unit with all necessary messages etc. -- if you have the clique at a particular point in time then (hopefully) everything else downstream becomes easier (maybe).

My reservation surrounding messages on edges comes from now having to store/restore not just the clique, but also its edges.  Furthermore, the edges must now be duplicated for this or that clique, and synchronization between two cliques becomes harder and harder to track.  Hence the knee jerk.  That said, there are limits to how "defensive" the coding patterns should be...  A step-by-step CSM is/would be a great developer's helper, but not a replacement for end-to-end testing.  Hence the relent to, just build it on edges and preemptively add NOTES as might seem relevant.

There is a looming refactoring here on tree messages, as you already know.  This refactoring has slipped into the new year and likely best suited to do part of the *treeinit* effort.  I will likely have to punt the message refactoring another couple of weeks given my immediate schedule, but super excited to get that part cleaned up!

> A clique will block waiting for a message to be received or send.

Agreed, that sounds like the right behaviour.

> Currently I use normal julia channels so you can just put and take from the channels without duplication needed.

Agreed on Julia `::Channel`, however, the duplication I referred to above is from a different use-case origin -- i.e. developer's helper.

While building the treeinit code (as it stands now) I was forced to add additional `::Condition` signals to the tree+CSM structure too :-P.  This has to do with cases where Siblings cannot run independent of each other any more, but need to share information through the parent, which in turn introduces the additional signaling requirements.

Affie, 2020-01-15:
To restate how I currently implement message storage:
- Messages are stored in the CSM container. 
    - up, down, upinit, downinit, etc (separate for debugging)
    - This should allow a clique to be solved in isolation, with a little work needed to emulate the message passing on the channels if you want to debug those states, or we can just skip the wait for message states completely.
- Messages are communicated with channels associated with edges. 
    - say you have tree with 3 cliques and 2 edges: 1 -> 2, 1 -> 3.
     - an up and down channel is created for each edge.
     - stored eg. `upMsgChan{Edge,Channel}(Edge(1,2) => Channel{BeliefMessage}(0)`

Affie, 2020-01-20:
> Messages are stored in the CSM container

The reason is messages are created at run time specific to the solve. If messages were to be stored in the clique, one would not be able to run 2 solves in parallel on the same tree (as currently implemented without a message `solveKey`). If that is even something we are interested in?

Affie, 2020-01-27:
@denann, I updated the summary a bit. I think the biggest issue on the message is `BallTreeDensity` field vs `val`, `bw` arrays. I used the arrays from EasyMessage in parametric. As I understand you `BallTreeDensity` is the better way to go to save on conversions? 
I missed that comment, I tried to find one message that could **work for both** and that could be **easily serialized**

My original idea was to use inheritance to allow for different messages since I couldn't pack it in `BallTreeDensisty`. We can go back to the parametric type idea also.

dehann, 2020-01-30:
I have been thinking about BeliefMessage types also, will look more closely.  I want to do best effort on BeliefMessage defintion and but then get to the meat of the tree message channels as fast possible, reduce the aperture there first.  Belief messages are not the biggest computational part at this stage, so okay if its a little slower.  I think its more important to get a common structure between the parametric and previous nonparametric CSM message passing.  That way a fix for either results in a fix for both solving methods, hence the hit rate goes up.  We can optimize stuff then.

dehann, 2020-02-03:
Haven't forgotten -- this is high on my priority list

dehann, 2020-02-06:
Slight change of tack here, this will be affected by #579  which will allow messages to encode the joint correlations beyond priors.  So, lets resolve 579 first then consolidate the messages -- PoC on 579 is actually pretty easy.

Affie, 2020-03-03:
Just to repeat it here. For #600 we will need additional information in the messages for covariances.

dehann, 2020-03-18:
> need additional information

Right, just to follow, which data elements are needed?

Affie, 2020-03-20:
#### Intermediate used for consolidation
Based on the currently used `TempBeliefMsg` and `EasyMessage`
- `val::Array{Float64,2}`
- `bw::Array{Float64,2}`
- `btd::BallTreeDensity`
- `inferdim::Float64`
- `manifolds::Tuple{Vararg{Symbol}}` . 
    - Occurred in `EasyMessage`
    - ~We have the variable available, so it should not be needed.~ It is needed, see next comment
    - DF from review comment: can't remember immediately but I do recall something about any code using TempBeliefMsg having to go back to a/the factor graph to getManifolds using the provided Symbol -- it was a schlep because it required a valid dfg object to be passed all the way down to that function...

If I'm not mistaken `BallTreeDensity` in `TempBeliefMessage` is made with `(getVal(vnd), getBW(vnd)[:,1], getSofttype(vnd).manifolds)` 
 - DF, there is a new constructor, e.g. `manikde!(getVal(vnd), getBW(vnd)[:,1], Pose2/Point/Pose3/Polar/etc)`
 - From `TempBeliefMessage` comment bellow DF, this also created large amounts of unnecessary memory allocation and repeat type conversion workload overhead, and think a chunk of that is still being done today.

The Belief Message
 - `status::CliqStatus`
    - New field needed for parametric combining belief and status in one channel 
 - `belief::Dict{Symbol, TreeBelief}`
    - This is similar to existing `TempBeliefMsg` and `EasyMessage`

- `struct TreeBelief`
    - [x] `TreeBelief` name, (can be `EasyMessage`, see note bellow)
    - [x] `val::Array{Float64,2}`
    - [x] `bw::Array{Float64,2}`
    - [ ] ~~`BallTreeDensisty`~~
    - [ ] ~~`TempBeliefMsg` (`BallTreeDensisty`) not needed as a separate message type.~~
    - [x] `inferdim::Float64`
    - [x] `manifolds::Tuple{Vararg{Symbol}}` 
    - [x] `softtype::T`

- `Struct LikelihoodMessage`
    - [x] `LikelihoodMessage` name 
    - [x] `status::CliqStatus`
    - [x] `belief::Dict{Symbol, TreeBelief}`
    - [x] `CliqStatus` as enumerated type

Note: I chose different names on purpose as not to confuse it with the current  `EasyMessage` and `TempBeliefMessage`. `TreeBelief` is essentially `EasyMessage` but with a Matrix for `bw` and not a vector (for now needed for parametric, related to #465).

Affie, 2020-03-20:
In:
```julia
struct LikelihoodMessage{T <: SamplableBelief} #<: Singleton
  status::CliqStatus
  variableOrder::Vector{Symbol}
  cliqueLikelihood::{T} # MvNormal for parametric
end
```
We still need a plan for:
- `inferdim`
- `softtype` (for manifold and variable dimension)

Currently `TreeBelief` is used/suggested as a consolidation bridge
- The name is not good, other names: `variableLikelihood`, `variableBelief`,  

**TreeBelief:**
```julia
struct TreeBelief{T <: InferenceVariable}
  val::Array{Float64,2}
  bw::Array{Float64,2}
  inferdim::Float64
  softtype::T
end
```
**Or add separate fields**
- `Dict{Symbol, Int)` - (variableLabel, inferdim) 
- `Dict{Symbol, InferenceVariable}` - (variableLabel, softtype)

dehann, 2020-03-25:
beliefs are consolidated to intermediate types (v0.11.0).  Work now is to consolidate message channels and functions that pass information around.  Will do this latter phase in v0.11.x.

Affie, 2020-04-07:
Parametric Solve Flow Example

- Signal
     - Status and
     - Likelihood message [optional]
     - Transmitted over a `Channel` (wire)
- `take!` block (wait) until data is available on the channel and then copies it to the relevant container.
- `put!` sends the signal over the `Channel`
- up: from leaves to root
- down: from root to leaves

Simplified "solve" CSM used in parametric
- Initialize cliques with subgraphs
- upsolve 
    - cliques waits for signals [`take!`]  from children.
    - children (starting at leaves) does upsolves and sends signal  [`put!`] with likelihood message to parents
         - Status can be upsolved/marginalized etc. 
         - Likelihood message can be saved for reuse (if clique is marginalized for example)
    - children waits [`take!`] for downsolve signal
    - parents receives signals from children and saves the likelihood message in clique/CSMC for later use and debugging.
   - when all signals are received in a parent it calculates the clique likelihood and sends signals further.
   - when the root receives its messages it does solve and triggers down [`put!`].
- downsolve 
    - all cliques except root are waitng on down signals
    - downsolves are computed and signals send to waiting children
    - signal from parent are stored and used in down calculation
    - continued to leaves where downsolve finishes   
- FG is updated with results from cliques after downsolve completes.

dehann, 2020-07-10:
#781 completes up message consolidation with exclusive storage location as `BTND.upMsgChannel::Channel{LikelihoodMessage}(1)` only.

dehann, 2020-07-17:
Up message consolidation was done before v0.13.0, and down message consolidation is targeted for release in v0.14.0

dehann, 2020-07-19:
Let's move milestone to IIF v0.15.0 for down message consolidation work, this is to allow rapid intro of JuliaRobotics/DistributedFactorGraphs.jl#536, so that we can close out #237

PR #884 (2020-09-09): fast-forward csm monitor task work
Hi @Affie please merge as soon you can.  Just a heads up that more changes are coming to CSM soon as part of the push to close #459 .
PR #908 (2020-09-18): #459 dwnMsg consolidation
closes #459
closed 2020-09-18 by dehann