JuliaRobotics/IncrementalInference.jl #465: [FEATURE] make VND.bw a 3 index array in preparation for parametric

dehann, 2019-12-05:
Probably a DFG issue, but starting it here.  Need is to store off-diagonal covariances while using the same VND data structure.

dehann, 2019-12-05:
cc @Affie

dehann, 2019-12-05:
look at what the performance characteristics would be.

Affie, 2019-12-06:
I did a quick comparison test.
Array  
0.026597 seconds (8 allocations: 22.888 MiB, 55.53% gc time)
Tuple  
0.028543 seconds (8 allocations: 22.888 MiB, 29.23% gc time)
Struct  
0.015862 seconds (8 allocations: 22.888 MiB, 23.56% gc time)
Mutable Struct  
0.036606 seconds (1.00 M allocations: 38.147 MiB, 37.92% gc time)
I will look at the options when I have a bit of time

Affie, 2019-12-09:
Starting of we are only working with one Gaussian mode. Therefore the covariance can fit in VND.bw,
I do not believe that an array is the easiest way forward. I propose that we wait with this changes that we might or might not need in the future. For instance, having a typed array might fit in a lot better when working with manifolds such as SE3. We can consider changing to types with a get/set index defined on them to make it easier to transition from the current usage.

dehann, 2019-12-16:
Hi thanks for taking a look!  Wait on accessors -- got it; and TypedArrays ... havent used or even heard of them myself, will go read up some more, thanks again.

Affie, 2020-01-06:
What I meant was; for example an array of type SE3 `Vector{SE3}` and not `Array{Float64,2}`

dehann, 2020-03-19:
see edit here: https://github.com/JuliaRobotics/IncrementalInference.jl/issues/465#issuecomment-563282068

closed 2020-03-19 by dehann