JuliaRobotics/IncrementalInference.jl #467: [FEATURE] Wrapper for parametric factor evaluation

dehann, 2019-12-05:
How would you use a normal IIF factor in parametric? example.

dehann, 2019-12-05:
cc @Affie

Affie, 2019-12-08:
Ok to implement cost directly
```julia
function (s::LinearConditional)(X1::Vector{Float64},
                                X2::Vector{Float64};
                                userdata::Union{Nothing,FactorMetadata}=nothing)
  # getFactorMean(s)
  meas = s.Z.μ
  # getFactorCov(s)
  σ = s.Z.σ
  res = [meas - (X2[1] - X1[1])]
  return (res/σ) .^ 2
end
```

dehann, 2019-12-26:
No unfortunately i don't think we should re-implement the costs as duplicates -- okay for dev just to get started, but would rather do this with a wrapper over the existing factor definitions.  I can help with that and is on my TODO list.

In the mean time see common reference here:
- https://www.juliarobotics.org/Caesar.jl/latest/concepts/adding_variables_factors/
- https://www.juliarobotics.org/Caesar.jl/latest/examples/basic_definingfactors/

Affie, 2020-02-19:
I've temporary continued with the re-implementing approach for simplicity. 
It looks like a wrapper can potentially work on all factors looked at except `Pose2Point2BearingRange`

dehann, 2020-02-19:
We can sort out the wrappers for dual nonparametric parametric use in IIF v0.9.x

dehann, 2020-07-19:
will resolve this after #459 is completed.  Only down message consolidation remains at this point.

dehann, 2020-12-28:
The parametric API will change a little, while nonparametric residual function API will change quite a lot.  Think it is feasible to consolidate these in IIF v0.19.  Just posting here for advance warning.

dehann, 2020-12-31:
WIP here:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/dfc235645ce0b8abc7d897ed0beea0d2e958d787/src/CalcFactor.jl#L22

dehann, 2021-01-03:
Need to figure out why this is not working (from #1101 ):
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/0469062bc1599222233801d75c6b546e5ae10ea5/src/ParametricUtils.jl#L115-L116

Affie, 2021-01-21:
Perhaps related to https://github.com/JuliaRobotics/IncrementalInference.jl/issues/467#issuecomment-753656505
But errors on `# DOES WORK`
```
ERROR: LoadError: MethodError: no method matching Float64(::ForwardDiff.Dual{ForwardDiff.Tag{IncrementalInference.var"#367#368"{LightDFG{SolverParams, DFGVariable, DFGFactor}, IncrementalInference.FlatVariables{Float64}}, Float64}, Float64, 12})
Closest candidates are:
  (::Type{T})(::Real, ::RoundingMode) where T<:AbstractFloat at rounding.jl:200
  (::Type{T})(::T) where T<:Number at boot.jl:760
  (::Type{T})(::AbstractChar) where T<:Union{AbstractChar, Number} at char.jl:50
  ...
Stacktrace:
  [1] convert(#unused#::Type{Float64}, x::ForwardDiff.Dual{ForwardDiff.Tag{IncrementalInference.var"#367#368"{LightDFG{SolverParams, DFGVariable, DFGFactor}, IncrementalInference.FlatVariables{Float64}}, Float64}, Float64, 12})
    @ Base ./number.jl:7
  [2] setindex!
    @ ./array.jl:841 [inlined]
  [3] macro expansion
    @ ./multidimensional.jl:903 [inlined]
  [4] macro expansion
    @ ./cartesian.jl:64 [inlined]
  [5] _unsafe_setindex!(::IndexLinear, ::Matrix{Float64}, ::Matrix{ForwardDiff.Dual{ForwardDiff.Tag{IncrementalInference.var"#367#368"{LightDFG{SolverParams, DFGVariable, DFGFactor}, IncrementalInference.FlatVariables{Float64}}, Float64}, Float64, 12}}, ::UnitRange{Int64}, ::UnitRange{Int64})
    @ Base ./multidimensional.jl:898
  [6] _setindex!
    @ ./multidimensional.jl:887 [inlined]
  [7] setindex!
    @ ./abstractarray.jl:1261 [inlined]
  [8] SE2(X::Vector{ForwardDiff.Dual{ForwardDiff.Tag{IncrementalInference.var"#367#368"{LightDFG{SolverParams, DFGVariable, DFGFactor}, IncrementalInference.FlatVariables{Float64}}, Float64}, Float64, 12}})
    @ TransformUtils ~/.julia/packages/TransformUtils/cFZOw/src/TransformUtils.jl:843
  [9] (::Pose2Pose2{FullNormal})(wXi::SubArray{ForwardDiff.Dual{ForwardDiff.Tag{IncrementalInference.var"#367#368"{LightDFG{SolverParams, DFGVariable, DFGFactor}, IncrementalInference.FlatVariables{Float64}}, Float64}, Float64, 12}, 1, Vector{ForwardDiff.Dual{ForwardDiff.Tag{IncrementalInference.var"#367#368"{LightDFG{SolverParams, DFGVariable, DFGFactor}, IncrementalInference.FlatVariables{Float64}}, Float64}, Float64, 12}}, Tuple{UnitRange{Int64}}, true}, wXj::SubArray{ForwardDiff.Dual{ForwardDiff.Tag{IncrementalInference.var"#367#368"{LightDFG{SolverParams, DFGVariable, DFGFactor}, IncrementalInference.FlatVariables{Float64}}, Float64}, Float64, 12}, 1, Vector{ForwardDiff.Dual{ForwardDiff.Tag{IncrementalInference.var"#367#368"{LightDFG{SolverParams, DFGVariable, DFGFactor}, IncrementalInference.FlatVariables{Float64}}, Float64}, Float64, 12}}, Tuple{UnitRange{Int64}}, true}; kwargs::Base.Iterators.Pairs{Union{}, Union{}, Tuple{}, NamedTuple{(), Tuple{}}})
    @ RoME ~/.julia/dev/RoME/src/factors/Pose2D.jl:46
 [10] (::Pose2Pose2{FullNormal})(wXi::SubArray{ForwardDiff.Dual{ForwardDiff.Tag{IncrementalInference.var"#367#368"{LightDFG{SolverParams, DFGVariable, DFGFactor}, IncrementalInference.FlatVariables{Float64}}, Float64}, Float64, 12}, 1, Vector{ForwardDiff.Dual{ForwardDiff.Tag{IncrementalInference.var"#367#368"{LightDFG{SolverParams, DFGVariable, DFGFactor}, IncrementalInference.FlatVariables{Float64}}, Float64}, Float64, 12}}, Tuple{UnitRange{Int64}}, true}, wXj::SubArray{ForwardDiff.Dual{ForwardDiff.Tag{IncrementalInference.var"#367#368"{LightDFG{SolverParams, DFGVariable, DFGFactor}, IncrementalInference.FlatVariables{Float64}}, Float64}, Float64, 12}, 1, Vector{ForwardDiff.Dual{ForwardDiff.Tag{IncrementalInference.var"#367#368"{LightDFG{SolverParams, DFGVariable, DFGFactor}, IncrementalInference.FlatVariables{Float64}}, Float64}, Float64, 12}}, Tuple{UnitRange{Int64}}, true})
    @ RoME ~/.julia/dev/RoME/src/factors/Pose2D.jl:44
 [11] _totalCost(fg::LightDFG{SolverParams, DFGVariable, DFGFactor}, flatvar::IncrementalInference.FlatVariables{Float64}, X::Vector{ForwardDiff.Dual{ForwardDiff.Tag{IncrementalInference.var"#367#368"{LightDFG{SolverParams, DFGVariable, DFGFactor}, IncrementalInference.FlatVariables{Float64}}, Float64}, Float64, 12}})
    @ IncrementalInference ~/.julia/dev/IncrementalInference/src/ParametricUtils.jl:116
 [12] #367
    @ ~/.julia/dev/IncrementalInference/src/ParametricUtils.jl:169 [inlined]
 [13] chunk_mode_gradient!(result::DiffResults.MutableDiffResult{1, Float64, Tuple{Vector{Float64}}}, f::IncrementalInference.var"#367#368"{LightDFG{SolverParams, DFGVariable, DFGFactor}, IncrementalInference.FlatVariables{Float64}}, x::Vector{Float64}, cfg::ForwardDiff.GradientConfig{ForwardDiff.Tag{IncrementalInference.var"#367#368"{LightDFG{SolverParams, DFGVariable, DFGFactor}, IncrementalInference.FlatVariables{Float64}}, Float64}, Float64, 12, Vector{ForwardDiff.Dual{ForwardDiff.Tag{IncrementalInference.var"#367#368"{LightDFG{SolverParams, DFGVariable, DFGFactor}, IncrementalInference.FlatVariables{Float64}}, Float64}, Float64, 12}}})
    @ ForwardDiff ~/.julia/packages/ForwardDiff/qTmqf/src/gradient.jl:142
 [14] gradient!
    @ ~/.julia/packages/ForwardDiff/qTmqf/src/gradient.jl:39 [inlined]
 [15] gradient!
    @ ~/.julia/packages/ForwardDiff/qTmqf/src/gradient.jl:35 [inlined]
 [16] (::NLSolversBase.var"#42#48"{IncrementalInference.var"#367#368"{LightDFG{SolverParams, DFGVariable, DFGFactor}, IncrementalInference.FlatVariables{Float64}}, ForwardDiff.GradientConfig{ForwardDiff.Tag{IncrementalInference.var"#367#368"{LightDFG{SolverParams, DFGVariable, DFGFactor}, IncrementalInference.FlatVariables{Float64}}, [... cut ...]
    @ Optim ~/.julia/packages/Optim/CK6Dn/src/multivariate/optimize/optimize.jl:33
 [20] solveFactorGraphParametric(fg::LightDFG{SolverParams, DFGVariable, DFGFactor}; solvekey::Symbol, autodiff::Symbol, algorithm::Type, algorithmkwargs::Tuple{}, options::Optim.Options{Float64, Nothing})
    @ IncrementalInference ~/.julia/dev/IncrementalInference/src/ParametricUtils.jl:170
 [21] solveFactorGraphParametric(fg::LightDFG{SolverParams, DFGVariable, DFGFactor})
    @ IncrementalInference ~/.julia/dev/IncrementalInference/src/ParametricUtils.jl:151
```

Affie, 2021-01-24:
> Need to figure out why this is not working (from #1101 ):
> https://github.com/JuliaRobotics/IncrementalInference.jl/blob/0469062bc1599222233801d75c6b546e5ae10ea5/src/ParametricUtils.jl#L115-L116
> 

Hi @dehann, I have parametric working with CalcFactor. However, the performance is terrable. Here is the @time results of a small fg:

New CalcFactor
# 2.234901 seconds (3.19 M allocations: 226.933 MiB, 2.25% gc time)

Old
# 0.005625 seconds (55.30 k allocations: 5.409 MiB)

I'll look at it some more, just wanted to let you know so you don't spend time on getting it running.

dehann, 2021-01-28:
This looks like type-stability issues?

Affie, 2021-01-28:
So changing to using returned residual to sort out the type instability yields results: 
New CalcFactor but with residual returned and not a parameter.
- 0.010857 seconds (87.80 k allocations: 9.533 MiB)
The extra allocations are from the extra layer caused by CalcFactor

Affie, 2021-01-28:
@dehann, here is the concept that I was thinking about, but I need your guidance if it looks likely to work and if you are ok with the maybe added complexity of giving users the ability to create 2 factor types?

As a possible longer-term solution, but for now, I think returning residual will work.

```julia

#The easy/basic factor function definitions
function (s::LinearRelative)(z,x1,x2) # where {M<:FactorMetadata,P<:Tuple,X<:AbstractVector}
  res = z - (x2 - x1)
end

#The complex factor function definitions
function (s::CalcFactor{<MyComplexFactor)(res, variables...) 
  #do something where you need other stuff...
  nothing
end

# then the default call for CalcFactor for non-parametric
function (s::CalcFactor)(variables...) 
  s.res .= s.factor(variables...)
  nothing
end

# and the default call for CalcFactorParametric for parametric
function (cfp::CalcFactorParametric)(variables...)
  res = cfp.calcfactor!(cfp.meas, variables...)
  return 0.5 * (res' * cfp.iΣ * res)
end
```

dehann, 2021-01-28:
Hi @Affie , thanks -- there is a lot to unpack here.  Think this is in the right ball park with some good but maybe a few pieces of this may have to change...  I want to very careful about communicating accurately here else we going to go in circles.  Give me a few hours to write clear (so-called "counter") examples.

> maybe added complexity of giving users the ability to create 2 factor types?

This is the part I'm most worried about (big 'no' from me unless you mean something specific).  Lets make sure we talking the same language here and forthcoming "counter" examples is probably best -- stay tuned...

dehann, 2021-01-28:
N.B. this a pointed technical discussion to try understand a far reaching multiple-dispatch issue -- no harm intended.  I don't know if any other hard-typed parametric and nonparametric libraries exist that can do both styles of inference using exactly the same member definitions as we are doing here. 

( The issue at hand is to use standard Julia autodiff tools while maintaining statically typed functions for all run-time, user defined factor types, including automated blends of `Mixtures`, `multihypo=`, `nullhypo=`, and access to advanced data management.  The user does not necessarily realize that all this comes from their single member definition.  Hence my drive below to constrain to a single but powerful API.)

### Never do this

```julia
function (s::CalcFactor)(variables...) 
  s.res .= s.factor(variables...)
  return sum(s.resl.^2) 
end
```
or
```julia
# then the default call for CalcFactor for non-parametric
function (s::CalcFactor)(variables...) 
  s.res .= s.factor(variables...)
  nothing
end
```

Specifically the part `s.factor(variables...)`, since that totally destroys any attempt of standardizing the API.  It's a quick fix for small/easy problems, but has no support for all the hard problems we have been facing for years.  So this cannot be used:
```julia
#The easy/basic factor function definitions
function (s__::LinearRelative)(z,x1,x2)
  res = z - (x2 - x1)
end
```

Because it is super important that users learn / use only ONE SINGLE API definition that works for all cases, that is `(s::CalcFactor{<:MyFactor})(z1, z2, x1,x2,x3,x4,...)`.  There is far too much history of wanting to get `metadata`, `smalldata`, `entry=>data`, `timestamps`, or needing to deal with `multihypo=`, or needing `.specialSampler`, etc. etc. etc. So I think for the short-medium term this will be an absolute requirement.  My concern is that Gaussian parametric only super-oversimplifies the problem which leads us away from solving the "hard" problem (i.e. non-Gaussian).  The practical cost of this is a standardized API based around `CalcFactor`, and best practice is maintain one standard throughout.  (think that is the fundamental issue we trying to figure out here).

Also remember there are also years of pain baked into `getSample(s::CalcFactor{<:MyFactor}, N)` which is not apparent in this residual factor write-up, but has same arguments.

Notice that the existing code forces us to use both NLsolve.jl and Optim.jl, which allows both `AbstractRelativeRoots` (more efficient) and `AbstractRelativeMinimize` (more flexible but slower) options.  Whatever we do here in 467 must maintain this requirement.  

Okay, so @Affie are you saying that this is not a workable solution for consolidated evaluation?
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/6ddd34628424d3a355bb75e37013b7aa92eec0d8/src/NumericalCalculations.jl#L61

------

### Should Standardize to Only This

```julia
#The standard factor function definition template
function (s::CalcFactor{<:MyFactor})(measurements..., variables...) 
  # maybe you need other stuff...
  # t1 = getTimestamp(s.metadata.dfgVariable[1])
  
  # maybe inline fails and you can avoid memory allocations by reusing
  # s.residual[1] = ... # some compute A
 
  # or you directly use your own internal
  res1 = ... # some compute A
  res2 = ... # more compute B

  # alternatively user is forcing in-place and does,  return s.residual
  return [res1; res2]  
end
```

This means we would do something like:
```julia
(s::CalcFactor{<:LinearRelative})(z,x1,x2) = z - (x2 - x1)
```

I really do not see a problem with using this later as a simple single standard that works for all known use-cases.

Are you saying @Affie there is absolutely no way in which you can get the type-stable dispatch to work (including todays fix #1138 ) using this later definition for both nonparametric and parametric?

Parametric should then be able to just use this for Malahanobis distance calculations (can do pseudo-Huber cost function later, etc later):
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/d06fe4cdf0ca8c7e19baaaccffd9d14a054380fe/src/ParametricUtils.jl#L103-L106

Also remember we will also do max-mixtures by consolidating parametric with `Mixture` -- that is a whole other dispatch party.

Finally, IIF v0.21.0 will likely again **bring a breaking change** on factor residual API, so that the residual function always returns `residual` (or `res`).  This is in contrast to current v0.20.0 format where factors always return `nothing`.

Affie, 2021-01-29:
### The "returning nothing" way
> Are you saying @Affie there is absolutely no way in which you can get the type-stable dispatch to work (including todays fix #1138 ) using this later definition for both nonparametric and parametric?

Not at all, this is the quickest and easiest way to get type stability, leave it up to julia to figure out the return type. I'm 100% happy with it. I started implementing it but got stuck on non-parametric that calls the function with the residual as a parameter and the difference that is then required between Minimize and Roots (as far as I understand).

### The "never do this" way
I'm not sure I follow your answer and if my example was clear enough, but it will break your ONE SINGLE API rule for factor definitions as there will be one for the simple (<:FunctorInferenceType) and one for complex (<:CalcFactor{<:FunctorInferenceType}) factor types.

Affie, 2021-01-29:
Perhaps we can get away with #1140

dehann, 2021-01-30:
> RE: returning the residual, "Not at all, this is the quickest and easiest way to get type stability, leave it up to julia to figure out the return type. I'm 100% happy with it."

That's great so we have consensus on that part!  I strongly agree with `leave it up to julia to figure out the return type`

dehann, 2021-01-30:
> I'm not sure I follow your answer and if my example was clear enough

I think I follow your example and if so, I'm trying to stop you from going down the path of creating an "easy" and "complex" interface.  I am very strongly against that idea.  Thats what I meant with "never do this way".

>  but it will break your ONE SINGLE API rule for factor definitions as there will be one for the simple (<:FunctorInferenceType) and one for complex (<:CalcFactor{<:FunctorInferenceType}) factor types.

This is what I'm trying to avoid and stop.  Let me look at 1140 and see if I can say anything more useful.  But just note, at this stage I would not merge anything on factors that breaks the "one API only requirement".

dehann, 2021-01-30:
With #1140 as a 'towards' consolidation it looks fine.  I know the existing parametric factor definitions follow the "easy" interface that still excludes the new `CalcFactor` API approach (https://github.com/JuliaRobotics/IncrementalInference.jl/issues/467#issuecomment-588110412).  The reason for 467 is to only end with one single API, hence we have to do away with the idea of "easy" and "complex" interface.  The logic shall from here on out just be "one API, that is as simple as possible, and `CalcFactor` looks like the best way to do it".

To illustrate again, I really cannot see what the 'complexity' is with a definition like this (which satisfies the "one api requirement"):
```julia
(s::CalcFactor{<:LinearRelative})(z, x1,x2) = z - (x2 - x1)
```

dehann, 2021-01-30:
Just to be clear, we have a project level requirement that there should be just one API for residual factor evaluation.  In the future there will be multiple algorithms and our history of using these factors in different applications **FORCES** us to create the headroom via `CalcFactor`.  No point in using Julia if we are manually writing factors for each dispatch case, and furthermore way way way too confusing when new users are trying to learn the library.  

This one API is now definitely an absolute requirement for the foreseeable future.  Sorry for pushing so hard on this, but let's just get it done and keep moving.

Affie, 2021-01-31:
>  I'm trying to stop you from going down the path of creating an "easy" and "complex" interface

I just wrote down an idea to enable factor reuse. It would still all be `CalcFactor`, but you can re-use the simple definitions in `CalcFactor` I don't know if it would ever be needed though. 
It's not something I looked at more than writing it down. Also not related to 1140

Affie, 2021-01-31:
Regarding #1140, I merged it as it can use `CalcFactor` as is (I still have to be tested in RoME though). The "return nothing" way.
 
The next step with parametric is to update RoME for Parametric CalcFactor and properly test it with #1140. 

We should decide if factors should return residual or change parameter and return nothing. 
- It might be easier with manifold integration to return the residual.
- Returning Nothing makes it harder for parametric to know the return type beforehand to pass in as a parameter. I don't know if the hack to always use the element type of the firs argument will hold.

dehann, 2021-02-04:
ping @Affie 

> We should decide if factors should return residual or change parameter and return nothing

Sorry if that has not been clear.  **Yes please always return the residual for both Roots and Minimize.**  This is the the new single API for factor residual functions.

> I don't know if the hack to always use the element type of the first argument will hold. 
 
I doubt this will hold for very long, but lets get the consolidation done and then worry about generalizing this part.  The main task item right now is ditching two factor definitions for different algorithms or uses.

dehann, 2021-02-06:
Hi @Affie , thanks for all the effort and 1147 -- think we can close this issue now right?

Affie, 2021-02-06:
I just want to do this before closing:
- [ ]  Update RoME and confirm its working.
- [ ] Remove old parametric factor functors.

closed 2021-02-06 by Affie