JuliaRobotics/IncrementalInference.jl #473: [MAINTENANCE] rename adjacency -> incidence for orderings

dehann, 2019-12-12:
compliments of @tonioteran 

https://github.com/JuliaRobotics/IncrementalInference.jl/blob/3d95fd14066f14f7c8638ad9d3eff9ca193294a8/src/FactorGraph01.jl#L1080

`Ind` Incidence is the rectagular edge matrix describing factors (rows) and variables (columns).

Laplacian is the square (literally) matrix:
```math
L = Ind' * Inc
```

Adjacency `A`

```math
L = D - A
```

where D is the diagonal degree matrix.

cc @GearsAD

dehann, 2019-12-12:
degree of variable is sum of column entries in `Ind`

Affie, 2020-01-06:
In graph theory, the incidence matrix is a matrix of vertices and edges (not vertices and vertices). https://en.wikipedia.org/wiki/Incidence_matrix#Graph_theory

dehann, 2020-01-16:
Okay, do you perhaps have a suggestion, we likely need something other than adjacency to be consistent with SLAM literature...

Affie, 2020-01-16:
Most accurate description I can find in the literature remains `biadjacency` matrix, from wikipedia and in some papers also: https://en.wikipedia.org/wiki/Adjacency_matrix#Of_a_bipartite_graph

tonioteran, 2020-01-16:
Why biadjacency?

Affie, 2020-01-16:
[image]

Affie, 2020-01-16:
[image]

tonioteran, 2020-01-16:
I understand the definition of the biadjacency, but I don't understand the rationale of using it. This is the same issue as in JuliaRobotics/DistributedFactorGraphs.jl#249.

We are doing edges and variables, not variables and variables. If variables and variables were being used, we would be getting square matrices, which we are not. For all intent and purposes, factors---even though vertices in the factor graph---are not variables. For reference, you can checkout Michael Kaess' iSAM2 paper, Figure 3, in which 3(a) is the specific matrix we're dealing with.

Affie, 2020-01-16:
I think I see where our misunderstanding is. `getAdjacencyMatrixSparse` works on the factor graph, so its a rectangular biadjacency matrix between variable nodes and factor nodes. Factors are not edges, so it can't be an incidence matrix.
On the other hand, with the Bayes net you would have an incidence matrix between edges (conditionals) and nodes (variables). 
Figure 3a is the measurement Jacobian related to the Gaussian only case.
`getAdjacencyMatrixSparse` is a basic graph operation returning a matrix of variables/factors with a 1 if there is an edge between the factor and variable, nothing more.

tonioteran, 2020-01-16:
`getAdjacencyMatrix[Sparse]` does indeed work on the factor graph, but it does not return neither a rectangular matrix, nor a rectangular biadjacency matrix. You can go ahead and run it:
```julia
fg = loadCanonicalFG_Kaess()
getAdjacencyMatrix(fg)
```
and one gets 
```julia
7×6 Array{Union{Nothing, Symbol},2}:
 nothing  :l1      :l2      :x1      :x2      :x3    
 :x1f1    nothing  nothing  :x1f1    nothing  nothing
 :x1l1f1  :x1l1f1  nothing  :x1l1f1  nothing  nothing
 :x1x2f1  nothing  nothing  :x1x2f1  :x1x2f1  nothing
 :x2l1f1  :x2l1f1  nothing  nothing  :x2l1f1  nothing
 :x2x3f1  nothing  nothing  nothing  :x2x3f1  :x2x3f1
 :x3l2f1  nothing  :x3l2f1  nothing  nothing  :x3l2f1
```
a rectangular matrix with variables on the columns and factors on the rows. The factors act as the constraints---or for all intent and purposes, as the edges---between our variables. If it were any sort of adjacency matrix, it would have to be square, i.e., (variables+factors) X (variables+factors) in the case of your biadjacency matrix.

What `getAdjacencyMatrix[Sparse]` returns is exactly what Figure 3(a) returns, just a boolean version of it without the actual numerical values; these boolean variables indicate the incidence between the variables in our factor graph, which are due to the factors or constraints of the system. That's why I had originally suggested the name `getMeasurementMatrix`, since the factors basically encode the measurements of the world that we're acquiring, but dehann thought it sounded too gtsam-y. Does that make sense?

Affie, 2020-01-17:
Here is a small example to illustrate why I feel using incidence is confusing when working with graphs:
```julia
fg = LightDFG{SolverParams}(params=SolverParams())
addVariable!(fg, :a, ContinuousScalar)
addVariable!(fg, :b, ContinuousScalar)
addFactor!(fg, [:a,:b], LinearConditional(Normal()))
```
[image]

```julia
julia> B = getAdjacencyMatrixSparse(fg)[1] |> Matrix
1×2 Array{Int64,2}:
 1  1

julia> A = LightGraphs.adjacency_matrix(fg.g) |> Matrix
3×3 Array{Int64,2}:
 0  0  1
 0  0  1
 1  1  0

julia> I = LightGraphs.incidence_matrix(fg.g) |> Matrix
3×2 Array{Int64,2}:
 1  0
 0  1
 1  1
```

dehann, 2020-02-02:
I think @Affie has won me over (this is nothing new, he's basically always right on these things).  Sorry @tonioteran .

Working from the current [Wiki definition for BiAdjacency](https://en.wikipedia.org/wiki/Adjacency_matrix#Of_a_bipartite_graph) and Johan's example above -- we need to also transpose the returned matrix... Check this out (according to example above):
```
B = [1;
        1]
Rv = [0 0; 0 0]  # two variables
Rf = [0] # one factor

biA = [Rv B; B' Rf]
```

Just as @Affie has been saying since https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/249#issuecomment-568104257 .  I need the hard example, laid out in fine detail though, and think this example from @Affie and references linked below should settle the discussion for future readers too.  Thanks everyone -- know its been a slog!  Solid group decisions are worth it though. 

### References Used
- https://en.wikipedia.org/wiki/Laplacian_matrix#Incidence_matrix
- https://en.wikipedia.org/wiki/Adjacency_matrix#Of_a_bipartite_graph
- https://github.com/JuliaRobotics/IncrementalInference.jl/issues/473#issuecomment-575250145
- https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/249#issuecomment-568343255

dehann, 2020-02-02:
closes JuliaRobotics/DistributedFactorGraphs.jl#249

dehann, 2020-02-27:
xref #592

dehann, 2020-03-09:
closed by #604

closed 2020-03-09 by dehann