JuliaRobotics/IncrementalInference.jl #476: [FEATURE] spyCliqMat(tree::BayesTree)

dehann, 2019-12-13:
for debugging in reverse -- see PoC hacks here:
https://github.com/JuliaRobotics/Caesar.jl/blob/784c455cf8bf244f7b570cbc0123f915d75d5dd4/examples/dev/BruteForceTreeOrderings.jl#L77

dehann, 2019-12-13:
Maybe this was already the plan, should we move the costing proxies into a Utils file of IIF as standard tools for checking the quality of the variable ordering -- perhaps helps people find more bugs?

I mean we should add:
https://github.com/JuliaRobotics/Caesar.jl/blob/784c455cf8bf244f7b570cbc0123f915d75d5dd4/examples/dev/BruteForceTreeOrderings.jl#L238-L298

to a new IncrementalInference/src/TreeCostUtils.jl file or something like that?

tonioteran, 2019-12-16:
I was just thinking about creating a IncrementalInference/src/AnalysisTools.jl file, but TreeCostUtils.jl would also work. Should I go ahead and do that?

dehann, 2019-12-16:
Hi yeah, that's a great idea -- either/or file names work.

closed 2020-02-27 by dehann