JuliaRobotics/IncrementalInference.jl #485: Deadlock solving graph using proposed release IIF 0.8.3 and DFG 0.5.2

GearsAD, 2019-12-17:
It looks like it's deadlocking on a test graph? This is a reproducible sample:

```
using Logging
using DistributedFactorGraphs
using IncrementalInference
logger = SimpleLogger(stdout, Logging.Debug)
global_logger(logger)
dfg = initfg() #LightDFG{SolverParams}(solverParams=SolverParams())
# Add some nodes.
v1 = addVariable!(dfg, :a, ContinuousScalar, labels = [:POSE])
v2 = addVariable!(dfg, : ContinuousScalar, labels = [:POSE])
v3 = addVariable!(dfg, :c, ContinuousScalar, labels = [:LANDMARK])
f1 = addFactor!(dfg, [:a; :b], LinearConditional(Normal(50.0,2.0)) )
f2 = addFactor!(dfg, [:b; :c], LinearConditional(Normal(50.0,2.0)) )
tree, smtasks = solveTree!(dfg)
```

Travis: https://travis-ci.org/JuliaRobotics/DistributedFactorGraphs.jl/jobs/626015919

dehann, 2019-12-17:
right, will have a look thanks!

dehann, 2019-12-17:
assuming these are all current master on DFG and IIF?

dehann, 2019-12-17:
also, if this fails so easy we should perhaps add this as standard test in IIF, not via loopback into special DFG downstream tests

dehann, 2019-12-17:
Oh, but this has no prior -- so the answers wont be correct in this test regardless.

dehann, 2019-12-17:
yes, the test fails by me too with CSM cycling indefinitely -- will need to resolve this before we can tag either DFG or IIF thanks

dehann, 2019-12-17:
This is likely a consequence of changes relating to #474.

GearsAD, 2019-12-18:
Yep cycle loopback isn't great for finding these, will add as test in IIF :+1:

dehann, 2019-12-18:
without prior, here is the current Bayes tree:
[image]

dehann, 2019-12-18:
this tree is wrong in that one of the factors (bcf1) looks to be missing.

dehann, 2019-12-18:
So this could also just be a visualization mistake, because if I check manually the factor is there:
[image]

dehann, 2019-12-18:
likely related to #461

dehann, 2019-12-18:
One of the problems here is likely the same thing as in #458 .

There is a separate visualization issue where leaf clique is not showing the factors that are in the clique -- will resolve that in a separate issue #489.

dehann, 2019-12-26:
test includes prior and visualization issue tracked separately, therefore closing this:
- https://github.com/JuliaRobotics/IncrementalInference.jl/blob/52b44d800073026ea687bd051e6659fc2cfc801c/test/testBasicCSM.jl#L28

closed 2019-12-26 by dehann