JuliaRobotics/IncrementalInference.jl #530: This looks like redundant code

Affie, 2020-01-10:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/708e62902104a8971a7cf1228abf01bbc2f0dac0/src/SolveTree01.jl#L1039-L1048

Affie, 2020-01-10:
It copies from `tree_` to `tree_`, should it not rather be from `treel`

Affie, 2020-01-10:
Also the direct indexing from one tree to another feels a bit risky as the index might not be the same?
Especially in the future where we might have a tree in the cloud.

dehann, 2020-01-16:
Oh dang, yeah that may be a bug.

> feels a bit risky as the index might not be the same?

Probably a bit risky, but relying on the tree building being deterministic given the same variable ordering.

> Especially in the future where we might

The better way to do this is actually modify the existing tree, rather than building a new one and copying values across.

> we might have a tree in the cloud

Yes, but perhaps keep that as a medium term goal for now (my opinion).

PR #537 (2020-01-16): bug fix for #530
fixes #530
closed 2020-01-16 by dehann