JuliaRobotics/IncrementalInference.jl #540: TreeClique index field and underlining graphs structure indexes

Affie, 2020-01-16:
### The `TreeClique` `index` field
```julia
mutable struct TreeClique
  index::Int  
  label::Symbol 
  data::Any
  attributes::Dict{String, Any} #The drawing attributes
end
```
#### Incrementally building new trees with clique reuse needs:
- remove clique method
- unique identifier for clique

Also useful for future distributed trees (cloud) based corresponding concept in DFG.

#### LightGraphs.jl as replacement for Graphs.jl
- for efficiency vectors of Integers are used for adjacency, we can extend it and write our own structure but then we have to maintain it separately.
- the Integers does not maintain any order but stored in a vector.
    - so iteration is easy clIds, ie 1:getNumCliqs(tree)
 
#### `index` field
- useful as an unique identifier such as a counter or an autogenerated GUID.
- do not use directly as graph index as it complicates keeping the index in sync with the graph
    - can use getIndex(clique::TreeClique) ** maybe good way to deprecate**.
- the corresponding concept in DFG is the label that is unique within a graph.
- we should probably rename it to avoid confusion. `cliqId`/`id`?

#### Plan is that:
- Integer indexes always refer to the index in the underlying graph.
- `cliq.index` or rather `cliq.id`/`cliq.cliqId` is a unique id within a tree.

However, cliq.index is very deeply embedded in CSM.
The alternative would be to keep the index field updated, but then another field is needed to uniquely identify a clique.

#### Proposed way forward:
- [ ] first merge parametric since its basically implemented there.
- [ ] deprecating with getIndex(clique::TreeClique) seems like the easiest way to get rid of all the `cliq.index` calls
- [ ] replace direct access based on `cliq.index` calls with graph index.
- [ ] add `deleteClique` methods

dehann, 2020-01-21:
ping, see edits above

Affie, 2020-08-02:
Consider using https://github.com/matbesancon/VertexSafeGraphs.jl as a quick replacement for Graphs.jl.

dehann, 2020-10-21:
Don't think we should do a quick replacement, let's pick a sensible one and do that as we are able.  Graphs.jl is archived, so its not going anywhere (there won't be breaking changes :-).

dehann, 2020-11-20:
xref #609

closed 2020-12-10 by Affie