JuliaRobotics/IncrementalInference.jl #551: Clique label field and label attribute is a bit confusing with getter/setter.

Affie, 2020-01-17:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/c6ad54287a9a2ad849498390601f45e142b098d8/src/JunctionTreeTypes.jl#L29-L34

dehann, 2020-01-21:
I think this was a result of two "musts":
- Graphs.ExVertex has `.label` -- used for unique ID for code, while
- GraphViz support in Graphs.jl expects `attributes["label"]` -- used for printing frontals and separators for humans

Affie, 2020-01-22:
> Graphs.ExVertex has .label -- used for unique ID for code, while

I think only `.index` is needed.

dehann, 2020-07-19:
Think this will get sorted out as part of deprecating Graphs.jl based `::BayesTree`.  Removed from CSM message and belief consolidation list.

dehann, 2020-12-26:
I think this was resolved with new `CliqueId{T}` approach.  @Affie can confirm?

closed 2020-12-26 by dehann