JuliaRobotics/IncrementalInference.jl #558: [DEV] What is Affie working on right now?

dehann, 2020-01-21:
Hi @Affie, I need some help understanding what you are doing at the moment please -- could you tick all that apply.  We need to stay a little synchronized here, otherwise it really throws of the rest of the team off:
- [ ] Paramatric proof of concept 2D linear?
- [ ] Parametric PoC manifold (non-linear) improvement?
- [ ] Parametric integration with existing factor definitions?
- [ ] Parametric integration or improvements to current CSMC data containers?
- [ ] Parametric PoC using existing BayesTree?
- [ ] Preparation on current Bayes tree so that we can slowly move away from Graphs.jl for the BayesTree specifically?
- [ ] Building a new MetaBayesTree?
- [ ] Standardizing message channels on the existing Bayes tree?
- [ ] Building a new message channels on the existing Bayes tree?
- [ ] Building separate message channels on a new Bayes tree that is not at all integrated with current Bayes tree?
- [ ] Standardizing or incorporating new fields in existing Belief message types?
- [ ] Building new message types that work on existing Bayes tree?
- [ ] Building new messages that only work on a new MetaBayesTree and not a all consolidated or integrated with existing code?

Please list any other things below?

PS, if any of these items above are not consolidated, how and where are we going to consolidate them?

PSS, Please edit with sub-bullet if any of these items are already finished? 

PSSS, if you working on a whole bunch of things at once (where none of them are quite done), we are going to have to do an all hands phone call to sequence the work better so that the team can follow along.

dehann, 2020-01-22:
As I understand the two most importing things you are blocking on me on are:
- [ ] How to do covariance calculation for Gaussian belief prop,
- [ ] Consolidation of the tree message Channels in Bayes tree.

dehann, 2020-01-22:
Haha, thanks enjoyed the "state of the nation" theme!  Thanks this actually really helps a lot with scoping work on my side.  Lets work to clear this list of stuff before we start anything new?

dehann, 2020-08-12:
CSM and ParametricCSM should be interchangeable, so that users can float between various Chapman-Kolmogorov solution strategies in development and use.

Affie, 2020-12-10:
I'm closing this as the only remaining issue is captured in #467

closed 2020-12-10 by Affie

dehann, 2020-12-10:
Thanks @Affie, that was an major pile of work.  Thanks for all the effort!