JuliaRobotics/IncrementalInference.jl #602: Tree init regression in buildup to DFG v0.6 IIF v0.9

dehann, 2020-02-27:
RoME `testBeehive2D_CliqByCliq.jl` started failing on a treeinit test.  Avoided by reverting to graph init, but definitely needs to be resolved.

dehann, 2020-02-27:
looks like numerical performance is lost even though tree is solving.

dehann, 2020-03-19:
also see #621

dehann, 2020-08-27:
Tree init sequencing might be failing.  In one solve of hex example it looked like cliques 2, 5, 6 were still busy with down init when cliques 1, 3, 4 finished down solving (this is based on looking at the color legend of the tree visualization while solving).

Affie, 2020-10-15:
Issue can be resolved with getSolverParams(fg).useMsgLikelihoods=true and take

dehann, 2020-10-15:
Thanks good to know thanks, slowing getting stuff resolved!

dehann, 2020-10-23:
suspect DFG v0.10.7, IIF v0.17.0, and RoME v0.10.0

Affie, 2020-10-23:
Update RoME tests to use  getSolverParams(fg).useMsgLikelihoods=true and IIF v0.17

Affie, 2020-10-26:
Test restored in https://github.com/JuliaRobotics/RoME.jl/pull/343

closed 2020-10-26 by Affie