JuliaRobotics/IncrementalInference.jl #605: Mixture prior produced unimodal belief

tipf, 2020-02-27:
Hi, I played around with the MixturePrior to represent a bi-modal GMM, but the resulting KDE looks unimodal. Is this a bug or is my understanding and/or usage of the MixturePrior wrong?

This is my minimal example:
```
using IncrementalInference, RoMEPlotting

# init graph
fg = initfg()

# add first variable
addVariable!(fg, :x0, ContinuousScalar)

# add bi-modal mixture prior
Prior0 = MixturePrior([Normal(-5.0,1.0); Normal(0.0,1.0)], Categorical([0.5;0.5]))
addFactor!(fg, [:x0], Prior0)

# solve
fg.solverParams.N = 1000;
solveTree!(fg)

# plot the results
RoMEPlotting.plotKDE(fg, :x0)
```

dehann, 2020-02-27:
Hi @tipf , thanks for posting!  Let me dive in here and make sure about whats going on.  From reading the your code example I would expect it to work even at N=50, and confident that this should (and had been) working. Something silly must have crept in somehow and we clearly missing a test case here.

BTW, There has been some major major refactoring under the hood.  We are in the final stages of preparing release IncrementalInference.jl v0.9.0 and your post has caught us in just time to fix before v0.9.0 goes out.  I will include this example as a standard test to make sure this remains solid going forward.  Just give me a day or two to put the fix and release together please, if that's not too much delay.  Thanks again.

tipf, 2020-02-28:
Thanks for the fast response @dehann, your schedule sounds perfectly fine.

dehann, 2020-02-29:
@tipf, see fix and new test in #613.  This will be part of v0.9.0 which will be coming soon (days) -- we are busy fixing a few other regressions last due to refactor on master and will do the stable release as soon performance is stable again
[image]

closed 2020-02-29 by dehann

tipf, 2020-03-02:
Dear @dehann, unfortunately, I get the same erroneous behavior with the new version 0.9.0 release. Should I update anything else?

```julia
julia> Pkg.status()
    Status `~/Workspace/Julia/Project.toml`
  [904591bb] IncrementalInference v0.9.0
  [238d586b] RoMEPlotting v0.1.1

julia> versioninfo()
Julia Version 1.3.1
Commit 2d5741174c (2019-12-30 21:36 UTC)
Platform Info:
  OS: Linux (x86_64-pc-linux-gnu)
  CPU: AMD Ryzen 7 3700X 8-Core Processor
  WORD_SIZE: 64
  LIBM: libopenlibm
  LLVM: libLLVM-6.0.1 (ORCJIT, znver1)
Environment:
  JULIA_NUM_THREADS = 8
```

PR #613 (2020-02-29): fix and test MixturePrior,
closes #605
commit d9f479b5 (2020-02-29): fix and test MixturePrior,

closes #605