JuliaRobotics/IncrementalInference.jl #674: CSM message passing push or pull model

dehann, 2020-04-04:
## Problem

Assuming inter-clique data is stored in CSMC (although does not change too much if stored on the tree edges), should tree message channels be stored in the sending or receiving CSMC?  E.g., should upward message channels be stored in child or parent CSMC?

### Related:
- #459 (first nonparametric/parametric consolidation effort)
- #673 (super solving)

### Biases

DF, I have a preference for storing message channels inside CSMCs rather than on tree edges, loosely discussed in #459 

## Considerations

- Historically tree initialization requires a dictionary of up messages message channels are stored in the parent.

dehann, 2020-04-04:
Current CSM code looks like `cliqSolverData().upInitMsgs` needs a dictionary from children upward messages
- Looks like push model -- i.e. stored in parent.

Affie, 2020-04-05:
Parametric uses `Channel`s to move data between cliques, therefore its a push/pull (`put!`/`take!`) model to/from the channel. 
eg. For up, children pushes to up channel and parents pulls, blocking until all messages are received. 
Currently no parent/child cliques are needed, only the channels. Therefore, the storing of channels in the tree and not CSMC. 

This method definitely made it easier to develop parametric, however it still doesn’t do initialization.

dehann, 2020-04-06:
> therefore its a push/pull (put!/take!) 

Right, so push/pull has double meaning and I meant something else:  where is the physical location of the message channels?  Either on the edges or in parent/child csmc or btnd, or separately in the tree?  This is part of #675 but still slightly different, and only comes into play of we choose to store the channels in either BTND or CSMC -- I think this is the same reason why you liked storing channels on the tree (linked from edges).

This issue specifically relates to the guiding principle for building CSM behaviour: is the status of a child clique (or message from child clique) something CSM must fetch from the child (physical location of channel), or did the OTHER CSM process push that information into a channel local to this CSM.  

Again, this might sound frivolous with channels stored separately in tree or edges, but that might be much more difficult to debug.  The current `recordcliqs` and `statehistory` step by step FSM plumbing in FunctionalStateMachine.jl assumes the channels are stored in BTND, which makes the step by step debugging much easier (https://github.com/JuliaRobotics/IncrementalInference.jl/issues/675#issuecomment-608977931).

dehann, 2020-04-16:
Decided to work towards a "pull" model, that is a CSM can only update fields inside or directly associated with that clique, and not update values outside or in another clique / foreign csmc.

PR #747 (2020-06-15): prep to convert initupmsg to pull model
work in progress on #459, and partly resolves #674
commit 8704e591 (2020-09-18): deprecated some old bad code, fix #674
closed 2020-04-16 by dehann