JuliaRobotics/IncrementalInference.jl #675: Store message channels in CSMC or tree edges

dehann, 2020-04-04:
Following from #459, we should decide if message channels between cliques are stored in either csmc or tree edges.

Related decisions:
- #674 (push or pull model)
- #673 (simultaneous solvers need separate channels)

dehann, 2020-04-04:
I have a preference for storing message channels inside CSMCs rather than on tree edges, loosely discussed in #459.

dehann, 2020-04-04:
functions like `treeProductUp` do not have access to CSMC, since this is a debugging function that occurs separate of solver (nominally post-inference to debug a weird internal result).

Affie, 2020-04-04:
We should maybe make the distinction between the message channels and the messages itself. 
Message storage options are in clique or CSMC. 
For me, the message channels only makes sense in the tree or cliques (with my preference in the tree as you know), not in CSMC.

dehann, 2020-04-06:
> (with my preference in the tree as you know), not in CSMC.

Ah, I had it that your preference was not in the tree, maybe make the tree read only?

Nonparametric is currently storing the message channels in the tree cliques (BTND):
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/master/src/JunctionTreeTypes.jl#L259-L262

Perhaps parametric is currently storing messages in either tree
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/99a0620fc2fdc8eb98c28667697188f47589af87/src/JunctionTreeTypes.jl#L23
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/master/src/JunctionTreeTypes.jl#L203-L204

Affie, 2020-04-06:
> Perhaps parametric is currently storing messages in either tree

see comment about making a distinction between channels and stored messages: https://github.com/JuliaRobotics/IncrementalInference.jl/issues/675#issuecomment-609015360

dehann, 2020-08-19:
Think we ended up deciding opposite to much of what was said earlier: that msgs must not be stored in csmc, but somewhere on the tree.  It's not super important whether issues are on the edges or in BTND.  Currently the implementation calls for all messages to be stored inside the BTND, and can eventually move to edges if needed.

Affie, 2020-08-25:
I find it a bit hard to follow the msg channels in BTND. I think I have it figured out, is this correct (hopefully it can help in the future if it is):
[image]

Affie, 2020-08-25:
For completeness here is the approach taken in parametric:
[image]

It looks like the biggest difference is every clique has its own down channel.

Affie, 2020-08-25:
This is what I thought was happening, and another possibility:
[image]

dehann, 2020-08-26:
While i consume the figures (thanks), I really don't see the hold up between msg channels on edges or in BTND:  somewhere in the code a CSM execution asks "GetUpwardChannel(csmc)::Channel{LikelihoodMessage}" and elsewhere another piece of code asks for "GetDownwardChannel(csmc)::Channel{LikelihoodMessage}" ... what does it matter where the channel references are stored, as long as csmc has that pointer?  And in both cases `csmc` does contain references to both the `::BayesTree` and all `::BTND`, so it really should not matter?

I'll work through the figures when i get a chance, perhaps there is something I'm missing.

Regardless, to aid consolidation for #459, I'd ask that we please store the messages in BTND since CSM is that much bigger than ParametricCSM.  It is much harder to change CSM than ParametricCSM.  The two structures should be identical and this seems to be a growing pain during consolidation.  The only reason this would be hard is if the "GetMsgChannel(csmc)" functions were not possible, and I think they are easily possible?

Affie, 2020-08-27:
In short:

- 1 is if I follow your "pull" definition, it shares a down channel and has unused channels - I'm still not sure if this is how it is/(will be)
- 2 and 3 are basically the same except for unused channels in the root (3)

dehann, 2020-08-28:
Hi @Affie, CSM is currently halfway between 1 and 3, and I'm working towards making it only be like 1.

dehann, 2020-10-20:
Hi @Affie , to bring this to a decision -- are we going to do messages in channels that are not stored in either BTND or CSMC, but rather a separate MessageChannels store in BayesTree?  If so then BayesTree will in future have to be able to work with subset of cliques (not critical at this stage).  

We also need a 100% determinstic way to ensure that step by step FSM sandbox repeat calls of of CSM states are exact replicates of the situations at that instant.  Since `csmc.tree` is deepcopied, I think that should be fine?

Affie, 2020-10-20:
I don't think it matters that much. The only requirement that I have is to have a down channel per child. This simplifies sequencing of the cliques and can be used in the future to only send a message to one of the child branches.

So I'm consolidating to use the edge messages as in 2 for now. As 1 only has one down channel. My previous function `refTreeMessageChannelsFromBTND!` actually uses 3 and not 1, as it uses incoming edges (so children)

> We also need a 100% determinstic way to ensure that step by step FSM sandbox repeat calls of of CSM states are exact replicates of the situations at that instant. Since csmc.tree is deepcopied, I think that should be fine?

Remember the channels are not buffered in take!, so technically the tree will not hold any state information if channels are stored in it. `waitforup` and `waitfordown` is dependant on the channels, but it's not likely to be debugged in separation as it depends on messages from other cliques. After that, the message is stored in the clique and can only be changed on the next cycle (up/down)

For me, the best of both solution at this stage is to make copies of the channels in the CSMC. 
so you would have an up-channels and a down-channels vector. That way you don't need to have copies of the parent and child cliques in your CSMC. As the rule is to only communicate through the up and down channels.

dehann, 2020-10-21:
agree 100% with only communicate through channels as hard requirement.

down channel each child sounds good too, no issues there.

did not connect the dots that channel length 0 means tree doesn't have state -- that's very good i think.  in old design i had to sometimes reset channels manually.  your approach sounds better.

Duplicating channels for debug restep with FSM should be fine, as long as that copying is not the place where things break.  if your happy i'd say hi for it.  or avoid if easy, whatever makes the most sense.

so design decision on this issue then is model 2 only (old language "pull model") with take-only channel length 0.  If you happy with that @Affie we can close this issue as soon as able.

dehann, 2020-10-21:
ie channels definely not in BTND.

Affie, 2020-10-22:
Ok, so we can reevaluate in the future:
If we remove the tree from CSMC and only keep a clique, then we can copy the channels from the tree to the CSMC instead of the full parent and child cliques. 
It works by reference so the only difference will be with debugging as deep-copies are made.

closed 2020-10-22 by Affie

dehann, 2020-10-22:
JIp, as long as the deepcopy point is not some weird place where multithreaded timing etc becomes hard to debug -- sometime nice to just have full deepcopy of the object as is.  Guess we can fix with atomics if that comes up.