JuliaRobotics/IncrementalInference.jl #681: CSM case trying to rebuild cliqSubFg after marginalized

dehann, 2020-04-15:
A mid-sequence clique `x62` is failing
[image]

## Log

```julia
┌ Info: 22:14:15.527 | 9---9| x62 @ null | 0., true ? :uprecycled => getCliqStatus(csmc.oldcliqdata)=downsolved
└ @ IncrementalInference /home/dehann/.julia/dev/IncrementalInference/src/CliqStateMachine.jl:19
┌ Info: 22:14:15.527 | 9---9| x62 @ null | 2, build subgraph syms=[:x62, :l16, :l41, :x63, :x59]
└ @ IncrementalInference /home/dehann/.julia/dev/IncrementalInference/src/CliqStateMachine.jl:19
┌ Info: 22:14:15.527 | 9---9| x62 @ null | 4, isCliqNull_StateMachine, downsolved, len(prnt)=1, csmc.incremental=true
└ @ IncrementalInference /home/dehann/.julia/dev/IncrementalInference/src/CliqStateMachine.jl:19
┌ Info: cliq 9, child 21 status is null, isready(initUpCh)=false.
└ @ IncrementalInference /home/dehann/.julia/dev/IncrementalInference/src/TreeBasedInitialization.jl:536
┌ Info: cliq 9, fetched all.
└ @ IncrementalInference /home/dehann/.julia/dev/IncrementalInference/src/TreeBasedInitialization.jl:542
┌ Info: 22:14:40.563 | 9---9| x62 @ null | 0b, checkChildrenAllUpRecycled_StateMachine -- length(chldr)=1, sum(count)=0
└ @ IncrementalInference /home/dehann/.julia/dev/IncrementalInference/src/CliqStateMachine.jl:19
┌ Info: 22:14:40.563 | 9---9| x62 @ null | 2, build subgraph syms=[:x62, :l16, :l41, :x63, :x59]
└ @ IncrementalInference /home/dehann/.julia/dev/IncrementalInference/src/CliqStateMachine.jl:19
```

## Stack Trace

```julia
Variable x62 already exists in destination graph!
Stacktrace:
 [1] error(::String) at ./error.jl:33
 [2] copyGraph!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::Array{Symbol,1}, ::Array{Symbol,1}; copyGraphMetadata::Bool, overwriteDest::Bool, deepcopyNodes::Bool, verbose::Bool) at /home/dehann/.julia/dev/DistributedFactorGraphs/src/services/AbstractDFG.jl:829
 [3] #deepcopyGraph!#43 at /home/dehann/.julia/dev/DistributedFactorGraphs/src/services/AbstractDFG.jl:872 [inlined]
 [4] deepcopyGraph! at /home/dehann/.julia/dev/DistributedFactorGraphs/src/services/AbstractDFG.jl:872 [inlined]
 [5] buildCliqSubgraph!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::IncrementalInference.TreeClique; solvable::Int64) at /home/dehann/.julia/dev/IncrementalInference/src/SubGraphFunctions.jl:74
 [6] buildCliqSubgraph! at /home/dehann/.julia/dev/IncrementalInference/src/SubGraphFunctions.jl:68 [inlined]
 [7] buildCliqSubgraph_StateMachine(::CliqStateMachineContainer{BayesTreeNodeData,LightDFG{SolverParams,DFGVariable,DFGFactor},LightDFG{SolverParams,DFGVariable,DFGFactor},BayesTree}) at /home/dehann/.julia/dev/IncrementalInference/src/CliqStateMachine.jl:662
 [8] (::StateMachine{CliqStateMachineContainer})(::CliqStateMachineContainer{BayesTreeNodeData,LightDFG{SolverParams,DFGVariable,DFGFactor},LightDFG{SolverParams,DFGVariable,DFGFactor},BayesTree}; breakafter::Function, verbose::Bool, iterlimit::Int64, recordhistory::Bool) at /home/dehann/.julia/packages/FunctionalStateMachine/jlnqL/src/StateMachine.jl:52
 [9] cliqInitSolveUpByStateMachine!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::BayesTree, ::IncrementalInference.TreeClique; N::Int64, oldcliqdata::BayesTreeNodeData, drawtree::Bool, show::Bool, incremental::Bool, limititers::Int64, upsolve::Bool, downsolve::Bool, recordhistory::Bool, delay::Bool, logger::Base.CoreLogging.SimpleLogger) at /home/dehann/.julia/dev/IncrementalInference/src/CliqStateMachine.jl:841
 [10] tryCliqStateMachineSolve!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::BayesTree, ::Int64; N::Int64, oldtree::BayesTree, drawtree::Bool, limititers::Int64, downsolve::Bool, incremental::Bool, delaycliqs::Array{Symbol,1}, recordcliqs::Array{Symbol,1}) at /home/dehann/.julia/dev/IncrementalInference/src/SolveTree01.jl:1286
 [11] (::IncrementalInference.var"#373#374"{BayesTree,Bool,Int64,Int64,Bool,Bool,Array{Symbol,1},Array{Symbol,1},LightDFG{SolverParams,DFGVariable,DFGFactor},BayesTree,Int64})() at ./task.jl:358
```

Affie, 2020-04-15:
Hi, I know where that comes from. It is part of the standardization for errors. You can call that with the `overwriteDest=true` keyword to force an update.

Affie, 2020-04-15:
It might actually show another bug though.

dehann, 2020-04-15:
Yeah, this is a pre-existing bug that the new subgraph stuff is showing.  If you look at the log portion in the first comment, you'll see it's trying to rebuild the subgraph twice due to interesting interaction of marginalization and incremental clique recycling.  I'm not keen on overwriting destination so rather fix CSM (or as close as possible at the moment).  I'm testing a local fix now...

PR #682 (2020-04-16): subgraph cleanup, bugfix,
closes #681
commit 2cd875e9 (2020-04-16): subgraph cleanup, bugfix,

closes #681
closed 2020-04-16 by dehann