JuliaRobotics/IncrementalInference.jl #699: Check rebuildFactorMetadata!

Affie, 2020-05-07:
I think its overwritng data

```julia
function rebuildFactorMetadata!(dfg::AbstractDFG, factor::DFGFactor)
  # Set up the neighbor data
  neighbors = map(vId->getVariable(dfg, vId), getNeighbors(dfg, factor))
  neighborUserData = map(v->getSolverData(v).softtype, neighbors)

  # Rebuilding the CCW
  ccw_new = getDefaultFactorData(dfg, neighbors, getFactorType(factor), multihypo=getSolverData(factor).multihypo)
  setSolverData!(factor, ccw_new)

  #... Copying neighbor data into the factor?
  for i in 1:Threads.nthreads()
    ccw_new.fnc.cpt[i].factormetadata.variableuserdata = deepcopy(neighborUserData)
  end

  return factor
end
```

```julia
function getDefaultFactorData(
      dfg::AbstractDFG,
      Xi::Vector{<:DFGVariable},
      usrfnc::T;
      multihypo::Vector{<:Real}=Float64[],
      nullhypo::Float64=0.0,
      threadmodel=SingleThreaded  )::GenericFunctionNodeData where
        {T <: Union{FunctorInferenceType, InferenceType}}
  #
  nullhypo != 0.0 ? error("nullhypo being renovated and not available in this new format yet.") : nothing
  # prepare multihypo particulars
  # storeMH::Vector{Float64} = multihypo == nothing ? Float64[] : [multihypo...]
  mhcat, nh = parseusermultihypo(multihypo, nullhypo)

  # allocate temporary state for convolutional operations (not stored)
  ccw = prepgenericconvolution(Xi, usrfnc, multihypo=mhcat, nullhypo=nh, threadmodel=threadmodel)

  # and the factor data itself
  data_ccw = FunctionNodeData{CommonConvWrapper{T}}(false, false, Int[], ccw, multihypo, ccw.certainhypo, 0)
  return data_ccw
end
```

```julia
function convert(
            ::Type{IncrementalInference.GenericFunctionNodeData{IncrementalInference.CommonConvWrapper{F}}},
            d::IncrementalInference.GenericFunctionNodeData{P} ) where {F <: FunctorInferenceType, P <: PackedInferenceType}
  #
  # TODO store threadmodel=MutliThreaded,SingleThreaded in persistence layer
  usrfnc = convert(F, d.fnc)
  # FIXME add proper nullhypo value
  mhcat, nh = parseusermultihypo(d.multihypo, 0.0)

  # TODO -- improve prepgenericconvolution for hypotheses and certainhypo field recovery when deserializing
  # reconstitute from stored data
  # FIXME, add threadmodel=threadmodel
  ccw = prepgenericconvolution(DFG.DFGVariable[], usrfnc, multihypo=mhcat, nullhypo=nh)
  ccw.certainhypo = d.certainhypo

  ret = FunctionNodeData{CommonConvWrapper{typeof(usrfnc)}}( d.eliminated, d.potentialused, d.edgeIDs, ccw, d.multihypo, d.certainhypo, d.solveInProgress)
  # error("what what $(ret.fnc.certainhypo)")
  return ret
end
```

dehann, 2020-05-07:
possible issues on maybe (incorrectly) overwriting these values:
- eliminated
- potentialUsed
- solveInProgress

Affie, 2020-08-27:
So far save/load does not preserve:
- [ ] solverData.eliminated
- [ ] solverData.potentialused
- [ ] solverData.solveInProgress
- [ ] solverData.nullhypo

dehann, 2020-08-27:
Hi thanks Johan, could you perhaps double check on `nullhypo` -- I thought I added that to GFND for serialization / persistence a few weeks back?  If it's not there we should definitely look to add that soon.

Affie, 2020-08-27:
Hi, yes it is in, it was just not serialized. It is just duplicated https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/477#issuecomment-681845488
If you can confirm that is correct.

PR #862 (2020-08-27): fix rebuildFactorMetadata!
close #699
commit 7246f1b7 (2020-08-27): fix rebuildFactorMetadata! (#862)

close #699
closed 2020-08-27 by Affie