JuliaRobotics/IncrementalInference.jl #714: Simple variable with priors does not calculate correctly

Affie, 2020-05-28:
On masters of DFG (v0.8) and IIF

Here are 2 basic examples that does not work correctly.

```julia
fg = initfg()
x = addVariable!(fg, :x0, ContinuousScalar)
p1 = addFactor!(fg, [:x0], Prior(Normal(30,3)))
p2 = addFactor!(fg, [:x0], Prior(Normal(-30,3)))
solveTree!(fg)
pkde = plotKDE(fg, [:x0])
```

```julia
fg = initfg()
x = addVariable!(fg, :x0, ContinuousScalar)
Prior0 = MixturePrior([Normal(-30.0,3); Normal(30,3)], Categorical([0.5;0.5]))
p0 = addFactor!(fg, [:x0], Prior0)
p1 = addFactor!(fg, [:x0], Prior(Normal(30,3)))
p2 = addFactor!(fg, [:x0], Prior(Normal(-30,3)))
solveTree!(fg)
pkde = plotKDE(fg, [:x0])
```

dehann, 2020-05-28:
Hi, thanks!  I will take a look.

Affie, 2020-05-29:
Maybe this one also helps
[image]

dehann, 2020-05-30:
oh, sorry im not following whats happening on this second image?

dehann, 2020-05-30:
Also please add the preamble you using to generate the plots -- these are done with `RoMEPlotting.plotKDE` and Plots.jl?  I'm assume you have local changes to combine Plots.jl and RoMEPlotting.jl ... so its bit hard to see where the second plot with all y traces is coming from?

Affie, 2020-05-30:
The second plot is the same as the first, just repeated a bunch of times in a loop. It’s just meant to show the range of the result. 

I struggle with RoMEPlotting to actually display and combine plots so i used it to generate the plot and then combined the plots with Plots.jl manually. So it’s not the output of the script ran once, but repeatedly.

dehann, 2020-06-03:
Hi @Affie, so this case is not quite as straight forward as "correct" or "incorrect".  What you are expecting as a parametric result in the middle should be viewed as a histogram (or summation) of all the cases shown in your second plot.  Statistically the combined event could be anywhere between the two narrow priors.

So a "bug" would be more if the product belief lies outside the two priors, or something like that.  On the one hand we would like a pure settled Gaussian mode precisely between the two prior values (as a quick parametric calculation would give).  The parasitics of the numerical methods used introduces the variation.

On the positive side, by forcing two priors so narrow and far apart, you are effectively saying that anything between them is fair game (think of it like a poorly conditioned matrix, or something analogous).

Performance should be much more repeatable if the two measurements (priors) are not taken to be so confident (narrow).  Are you sure that at least one of the two priors are not overconfident?  See example below.

Also, the second example with third mixture prior is a very different animal.  The mixture prior is a summation of two modes.  The first example is a product of two Gaussian priors.

## Stochastic Example:
```julia
using IncrementalInference, RoMEPlotting

fg = initfg()
addVariable!(fg, :x0, ContinuousScalar)
addFactor!(fg, [:x0], Prior(Normal(30,20)))
addFactor!(fg, [:x0], Prior(Normal(30,-20)))

ARR = BallTreeDensity[]
[(solveTree!(fg); push!(ARR,getKDE(fg, :x0))) for i in 1:30]
plotKDE(ARR) # plot3
sms = hcat((rand.(ARR, 1000))...)
Gadfly.plot(x=sms, Geom.hist) # plot4
```

[image]

[image]

## Parametric

Analytical product is pretty close to the aggregate samples from the 30 products above:
```julia
pr(x) = exp((-x^2-900)/400)/800/pi

using Colors

Gadfly.plot(
        Gadfly.layer(x=-50:0.1:50, y=620*pr.(-50:0.1:50), Geom.line, Theme(default_color=colorant"red")),
        Gadfly.layer(x=sms, Geom.histogram(density=true)) 
)
```

Affie, 2020-06-03:
Thanks for the thorough explanation. 
There are 2 reasons why I looked at it, I wanted to confirm the covariance is calculated correctly for the parametric solves, and I thought it might be the cause of the numerical inaccuracies I'm seeing in larger solutions.
It looks like there is good agreement on the examples with the larger variance, so I think parametric works as it should. 
It's just the tight variance that gives a different result. Sorry for the unclear images previously, your histograms gives a much clearer picture. 

Just for clarity, here are the histograms based on your code examples but with the tight (3) variance. 
[image]

- Is there any parameter to tune to get less numerical parasitics, even if it comes at a cost of more computation time?  

- Regarding the second example. Is there a way to get 2 confident measurements to rather produce 2 modes rather than smear to somewhere in between without the miltihypo factor.

dehann, 2020-06-05:
I'd have to look more closely, but in the meantime:

> Is there any parameter to tune to get less numerical parasitics, even if it comes at a cost of more computation time?

try `getSolverParams(fg).N = 300` or something like that.

> Regarding the second example. Is there a way to get 2 confident measurements to rather produce 2 modes rather than smear to somewhere in between without the miltihypo factor.

If I understand you correctly, then using only the mixture prior you added in the second example should be what you want (without the unimodal priors).

Affie, 2021-02-27:
I think I'm still running into this issue (the effects of) with multihypo, it should not happen after #1010, so I'll re-evaluate after that is in.

Affie, 2021-10-06:
Bearing only issues (https://github.com/JuliaRobotics/RoME.jl/issues/344) possibly relates to this issue

dehann, 2026-03-04:
Ah, so we should be careful about which pathological cases we are looking at.  I'd say in this thread we stay with the existing title where the product between two "over-confident" priors will still produce a single posterior mode between them.

> f the "signal" evidence drops below the "noise" threshold, the conflicting measurements will naturally split into 2 modes.

This is likely to show up in a more enhanced cases, e.g.:
- if n times m modes result in cases of additional modes which may have very small weights.  More precise, 2 modes times 2 modes, where 1A lines up well with 2A, but 1B and 2B are both far outside confidence of any of the other modes -- this implies there are four modes in the posterior with only one "heavy" mode while 3 low weight modes can be lost in a stochastic process and/or waste computational resources in a brute force computation that exhausts all modes via naive search methods.