JuliaRobotics/IncrementalInference.jl #716: solveTree! stucks in Juno debugger

andreyzhitnikov, 2020-05-29:
Hi, 
 solveTree!  function always stucks when run in Juno debugger.

dehann, 2020-05-29:
HI @andreyzhitnikov, it might be due to the size of the computation -- we usually debug lower down calls individually.  Do you perhaps have a short example of whats failing?

andreyzhitnikov, 2020-05-29:
fg = initfg()
addVariable!(fg, :x0, Pose2)
addFactor!(fg, [:x0], PriorPose2( MvNormal([10; 10; pi/6.0], Matrix(Diagonal([0.1;0.1;0.05].^2)))))
sym = Symbol("l0")
addVariable!(fg, sym, Point2)
 @show isInitialized(fg, sym)
addFactor!(fg,[sym], PriorPoint2(MvNormal([1; 1], Matrix(Diagonal([0.3;0.3].^2)))))
@show getSolverParams(fg)
# ensureAllInitialized!(fg)
getSolverParams(fg).dbg = true
# list vertices in fg
@show ls(fg)

## Solve graph
tree, smt, hist= solveTree!(fg, maxparallel=1)

andreyzhitnikov, 2020-05-29:
When I run in REPL everything is fine. But in Juno debugger it stucks on 
do intit of x0
[image]

andreyzhitnikov, 2020-05-29:
It works when I'm setting graphinit option to off. 
opt = getSolverParams(fg)
opt.treeinit = true
opt.graphinit = false

andreyzhitnikov, 2020-05-29:
It happens also if I enter to this function with Debugger.jl. However, when running with REPL everything works fine
[image]

dehann, 2020-05-30:
I think its just loading too much at once and you are seeing the Debugger working its way through slowly.  First thing I'd suggest is to rather try this method on Juno command line:
```julia
Juno.@enter solveTree!(fg)
```
which can be stepped through more easily:
[image]

Also note, I think you might have misunderstood the use of maxparallel.  I have since change that in #717.  It is used to prevent very dense factor graphs (albeit an artificial constraint).  Multiprocess or threading is done differently, if that is how you understood it...  (apologies for the ambiguity).

A last note (which I will start doing more in the docs is simplified:
```julia
using LinearAlgebra

cov = diagm([0.1;0.1;0.01].^2)
```

dehann, 2020-05-30:
On this: https://github.com/JuliaRobotics/IncrementalInference.jl/issues/716#issuecomment-636111844

please note that treeinit is experimental and likely won't produce satisfactory results -- you can follow the progress here #602.  Best to use `graphinit` or `initManual!(fg, :x0,...)` interfaces at this time.

closed 2020-06-03 by dehann