JuliaRobotics/IncrementalInference.jl #744: Undo CliqStatus back to an enum

dehann, 2020-06-15:
Had to make it ::Symbol like older mm CSM, as part of #459 consolidation effort

dehann, 2020-07-09:
make sure `printHistoryLine` preserves the status name and not just an Int.

Affie, 2020-10-01:
FIY:
```julia
julia> @enum CliqStatus NULL INITIALIZED UPSOLVED MARGINALIZED DOWNSOLVED UPRECYCLED ERROR_STATUS

julia> st = ERROR_STATUS
ERROR_STATUS::CliqStatus = 6

julia> println(st);
ERROR_STATUS

julia> @show(st);
st = ERROR_STATUS

julia> display(st);
ERROR_STATUS::CliqStatus = 6
```

dehann, 2020-10-06:
Sounds good, lets get v0.16.0 out and then add this as a further part in CSM consolidation in v0.17.0?

dehann, 2020-10-18:
Can also go for v0.18.0 if that is better

Affie, 2020-10-19:
It shouldn't be too much effort, so we can still target v0.17

PR #988 (2020-10-20): Maintenance on take and Restore CliqueStatus enum
- CSM_X -> CSM
- Restore clique status as enum close  #744
- Cleanup TreeBasedInitialization
- Cleanup LinearConditional -> LinearRelative
-  isCliqMarginalizedFromVars deprecate close #962
closed 2020-10-20 by Affie