JuliaRobotics/IncrementalInference.jl #753: add watchdog to auto-stop dormant CSM tasks

dehann, 2020-06-20:
If something goes wrong and there is no activity anywhere on the tree (all CSMs are stuck or closed out) then an autostop on all tasks  -- perhaps self terminate on watchdog -- could be very useful, rather than loosing the entire Julia context.

Affie, 2020-06-20:
I did some thing like this in parametric, but can’t find the commit now. It was the reason behind the error propagation messages.

dehann, 2020-06-26:
One idea is perhaps to create a single point in BayesTree and all CSM tasks should reset their counters there until one exceeds and then the CSM itself throws a `WatchdogException` to terminate the processing -- little hard to design properly actually.

Affie, 2020-06-26:
I was thinking in the lines of a "per state" watchdog timer and not per clique, but yes, it's not straightforward.

dehann, 2021-02-25:
this was done with `=timeout`.

closed 2021-02-25 by dehann