JuliaRobotics/IncrementalInference.jl #754: CSM endless cycle fail case

dehann, 2020-06-20:
Found a easy fail case on CSM, relates to existing test:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/03d231d63fad0f7b2ea897c328a657753ca8d7c2/test/testBasicTreeInit.jl#L9

### MWE

Clique 5 and 6 solves cycle endlessly if the following clique solve sequence is used:

```julia
fg = generateCanonicalFG_CaesarRing1D(graphinit=false)
getSolverParams(fg).graphinit = false
getSolverParams(fg).limititers = 50
solveCliq!(fg, tree, :x5, async=true)  # , recordcliq=true
solveCliq!(fg, tree, :x3, async=true)  # , recordcliq=true
```

dehann, 2020-09-19:
also added as a recent new test in IIF, testBasicTreeInit.jl

Affie, 2020-09-22:
Another basic endless cycle fail case (after #459)
```julia
using IncrementalInference
fg = generateCanonicalFG_lineStep(5; 
                                  poseEvery=1, 
                                  landmarkEvery=5, 
                                  posePriorsAt=[0,2], 
                                  sightDistance=4,
                                  solverParams=SolverParams(algorithms=[:default, :parametric]))
                                  
getSolverParams(fg).graphinit = false
getSolverParams(fg).treeinit = true
smtasks = Task[]
tree, smt, hist = solveTree!(fg; smtasks, verbose=true, timeout=50, recordcliqs=ls(fg));
```

Affie, 2020-09-23:
[image]

dehann, 2020-09-25:
I'm going to debug a little by adding:
```julia
fid = open(joinLogPath(fg,"csmVerbose.log", "w")
solveTree!(...;  verbosefid=fid, verbose=true, ...)

# after finished
close(fid)

open(joinLogPath(fg, "csmLogicalReconstructMax.log"),"w") do io
  IIF.reconstructCSMHistoryLogical(getLogPath(fg), fid=io)
end
```

dehann, 2020-09-25:
Here it is:
[csmLogicalReconstructMax.log](https://github.com/JuliaRobotics/IncrementalInference.jl/files/5280411/csmLogicalReconstructMax.log)

dehann, 2020-09-25:
So this is the loop on (1):
```
trafficRe null
maybeNeed null
determine null
blockUnti null
```

While Children `(2)=>init`, and `(3)=>null`

First question for me is why does `blockUntilChildrenHaveStatus` not stop the loop, since one child is `null`...

dehann, 2020-09-25:
We can also look at the log on (1):
[log.txt](https://github.com/JuliaRobotics/IncrementalInference.jl/files/5280450/log.txt)

dehann, 2020-09-25:
Okay, so log shows something else, it says `(3)=>needdownmsg`:

```julia
┌ Info: 23:22:02.713 | 1---1| x2 @ null | 4e, blockUntilChildrenHaveStatus_StateMachine, maybe wait cliq=2, child status=upsolved.
└ @ IncrementalInference /home/dehann/.julia/dev/IncrementalInference/src/CliqStateMachine.jl:63
┌ Info: 23:22:02.713 | 1---1| x2 @ null | 4e, blockUntilChildrenHaveStatus_StateMachine, maybe wait cliq=3, child status=needdownmsg.
└ @ IncrementalInference /home/dehann/.julia/dev/IncrementalInference/src/CliqStateMachine.jl:63
┌ Info: 23:22:02.721 | 1---1| x2 @ null | 4b, trafficRedirectConsolidate459_StateMachine, cliqst=null
└ @ IncrementalInference /home/dehann/.julia/dev/IncrementalInference/src/CliqStateMachine.jl:63
```

dehann, 2020-09-25:
Yup, here is the transition that sets off the cycle in 1 (the last step from neighboring CSMs):
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/3a410316a0804ecee5accc3fc53ea0dfe32fd90d/src/CliqStateMachine.jl#L1150

So now we can look at how to resolve the process inside CSM (1) only with this pre-knowledge...

dehann, 2020-09-25:
So we need a way to redirect the loop (https://github.com/JuliaRobotics/IncrementalInference.jl/issues/754#issuecomment-698701536) towards one of the `slow` steps  -- either:
- `slowIfChildrenNotUpsolved_StateMachine`, or
- `slowWhileInit_StateMachine`

dehann, 2020-09-25:
Lets start with `slowWhileInit_StateMachine` as the most promising.

https://github.com/JuliaRobotics/IncrementalInference.jl/blob/3a410316a0804ecee5accc3fc53ea0dfe32fd90d/src/CliqStateMachine.jl#L740-L751

Yup, that looks pretty good.  The question now is which of the cycle member should divert out in this case.  Best is to read all the code in those 4 cycle functions and see which part is closest to the current case...

dehann, 2020-09-25:
So the first major thing to note is that there is currently zero exit points in this loop that would have gone to our desired `slowWhileInit_StateMachine` -- so that is a big deal.  This means we are not going to just tweak existing directions, we actually have to add a new transition edge in CSM.  

The question is where.  Two spots look the most promising:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/3a410316a0804ecee5accc3fc53ea0dfe32fd90d/src/CliqStateMachine.jl#L1077

or 
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/3a410316a0804ecee5accc3fc53ea0dfe32fd90d/src/CliqStateMachine.jl#L1226

To narrow this down, I'm going to look at what do each of these next states do:
- `maybeNeedDwnMsg_StateMachine` -- *"If all children (then also escalate to) :needdownmsgs and block until sibling status."*
- `blockUntilChildrenHaveStatus_StateMachine` -- *"Block until all children have a csm status, using `fetch---Condition` model."*

So the first one looks most relevant in this case, with the failure here that its not ALL CHILDREN leading to a BLOCK, but rather only some children are not inducing a SLOW.

I don't think the second option `blockUntil` is relevant in this case, because that was precisely the last step from (3) which set off this loop -- i.e. once all children had a status (https://github.com/JuliaRobotics/IncrementalInference.jl/issues/754#issuecomment-698702930) then it was no longer appropriate to block.

dehann, 2020-09-25:
Okay, so lets add something to change this transtion:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/3a410316a0804ecee5accc3fc53ea0dfe32fd90d/src/CliqStateMachine.jl#L1077

dehann, 2020-09-25:
If we change to:
  # Some traffic direction
  if cliqst == :null
    if doAnyChildrenNeedDwnMsg(csmc.tree, csmc.cliq)
      # go to 7e (#754)
      return slowWhileInit_StateMachine
    end

    # go to 4d
    return maybeNeedDwnMsg_StateMachine
  end

Okay, that fixed one problem and started a new one, now the leaf clique (4) runs away to the `iterlimit`:
[image]

dehann, 2020-09-25:
All right, here we go again:
[csmLogicalReconstructMax.log](https://github.com/JuliaRobotics/IncrementalInference.jl/files/5280635/csmLogicalReconstructMax.log)

The new cycle is much longer and might require a brief interaction with the parent clique to progress:
```
Leaf clique (4)

64m rmUpLikel need   |||   79m rmUpLikel need 
65m checkUpso need   |||   80m checkUpso need 
66m trafficRe need   |||   81m trafficRe need 
67m doAllSibl need   |||   82m doAllSibl need 
68m determine need   |||   83m determine need 
69m dwnInitSi need   |||   84m dwnInitSi need 
70m attemptDo need   |||   85m attemptDo need 
73m rmMsgLike need   |||   88m rmMsgLike need 
74m decideUpM need   |||   89m decideUpM need 
75m attemptCl need   |||   90m attemptCl need 
76m prepInitU need   |||   91m prepInitU need 
77m tryInitCl need   |||   92m tryInitCl need 

=============================================
Interlaced then (3) the parent does

58m slowWhile init

71m prepInitU init
72m tryInitCl init

78m slowWhile init

86m prepInitU init
87m tryInitCl init

93m slowWhile init

95m prepInitU init
96m tryInitCl init

97m slowWhile init
```

dehann, 2020-09-25:
The easiest thing now perhaps is to compare with Meld the different between this new solve attempt and the previous one and see where the two diverge:

[image]

dehann, 2020-09-25:
The easiest thing now perhaps is to compare with Meld the differences between the previous solve attempt (left) and this new (right) and see where the two diverge:

[image]

dehann, 2020-09-25:
We should definitely add this test case above to the IIF unit tests once we have it the issue sorted out.

dehann, 2020-09-25:
Hi @Affie , its getting late here -- we can catch up in the morning a debug for a bit.  Day will be a bit tight with work and presentation in the afternoon, so see how far you can get and I will try fill in again as soon I get a chance.

dehann, 2020-09-25:
ah, one more idea before going too far:  earlier i added a new transition to 'slowWhileInut' because that cycle had no such direct transition;  it's probably worth checking for an indirect path and where is the closest point in the cycle to redirect to that indirect path that leads to 'slowWhileInit'.

Affie, 2020-09-27:
Another, maybe related case:
fg = generateCanonicalFG_Circle(8;graphinit=false)

[image]

Affie, 2020-09-28:
Another endless cycle can happen in `determineCliqNeedDownMsg <---> dwnInitSyblingWaitOrder`
[image]

dehann, 2020-09-29:
Thanks, keep posting any such fail cases here, and we should add as many we can to the standard unit testing.

dehann, 2020-09-30:
Picking up the story from https://github.com/JuliaRobotics/IncrementalInference.jl/issues/754#issuecomment-698723186

The diff shows that clique 3 at sequence step 40 followed the new edge as entry into the new cycle.  We also see that the previous CSM blocked on Siblings.  We therefore move the new segment of code into `maybeNeedDwn`, see f38cfbc which fixed the problem:
[image]

dehann, 2020-09-30:
On `Circle(8,...)` case...

```julia
julia> printCliqHistorySequential(hists, [8=>20:33; 4=>31:34])
4.31   04:05:48.385 tryInitCliq          | needdownmsg P 2initialized C 8needdownmsg |S| 6upsolved 
8.20   04:05:48.406 trafficRedirectCon   | needdownmsg P 4initialized C ---- |S| 
8.21   04:05:48.412 doAllSiblingsNeedD   | needdownmsg P 4initialized C ---- |S| 
8.22   04:05:48.417 determineCliqNeedD   | needdownmsg P 4initialized C ---- |S| 
8.23   04:05:48.422 dwnInitSiblingWait   | needdownmsg P 4initialized C ---- |S| 
8.24   04:05:48.428 attemptDownInit      | needdownmsg P 4initialized C ---- |S| 
8.25   04:05:48.474 rmMsgLikelihoodsAf   | needdownmsg P 4initialized C ---- |S| 
8.26   04:05:48.479 decideUpMsgOrInit    | needdownmsg P 4initialized C ---- |S| 
8.27   04:05:48.484 attemptCliqInitUp    | needdownmsg P 4initialized C ---- |S| 
8.28   04:05:48.489 prepInitUp           | needdownmsg P 4initialized C ---- |S| 
8.29   04:05:48.496 tryInitCliq          | needdownmsg P 4initialized C ---- |S| 
4.32   04:05:48.501 slowWhileInit        | initialized P 2initialized C 8needdownmsg |S| 6upsolved 
8.30   04:05:49.035 rmUpLikeliSaveSubF   | needdownmsg P 4initialized C ---- |S| 
8.31   04:05:49.04  checkUpsolveFinish   | needdownmsg P 4initialized C ---- |S| 

8.32   04:05:49.045 trafficRedirectCon   | needdownmsg P 4initialized C ---- |S| 
8.33   04:05:49.05  doAllSiblingsNeedD   | needdownmsg P 4initialized C ---- |S| 
4.33   04:05:49.096 prepInitUp           | initialized P 2initialized C 8needdownmsg |S| 6upsolved 
4.34   04:05:49.1   tryInitCliq          | initialized P 2initialized C 8needdownmsg |S| 6upsolved 
```

Both cliques 4 and 8 should hit a stopper while parent is `initialized`.

dehann, 2020-09-30:
Just to record the debugging process.  I'm looking at the 4.31 to 4.32 transition (parent of 8) going from `:needdownmsg` to `:initialized`.  The function `tryCliqInit` does appear to send a down message with `status=:initialized` which triggers child event 8.20.

At this point it looks like the child (8) is not taking that message and doing the required down init to get to status `:upsolved`...?

dehann, 2020-09-30:
btw the parent being upsolved first is an error #924

dehann, 2020-09-30:
```
8.24   04:05:48.428 attemptDownInit      | needdownmsg P 4initialized C ---- |S| 
```

seems like the most likely point where the down init should have occurred

dehann, 2020-09-30:
So this seems a little weird:
```julia
┌ Info: 04:05:48.433 | 8---8| x4 @ needdownmsg | 8e.ii, attemptDownInit_StateMachine, do cliq init down dwinmsgs=Symbol[]
└ @ IncrementalInference /home/dehann/.julia/dev/IncrementalInference/src/CSMCommon_Consolidate.jl:21
```

dehann, 2020-09-30:
Ah, I tried it again with `.dbg = true` and that seemed to introduce enough delay somewhere and in a way that allows the solution to complete -- we can use this as reference to see where the CSM went astray:

[image]

Verbose reconstruction of csmLogical:  [csmLogicalReconstructMax.log](https://github.com/JuliaRobotics/IncrementalInference.jl/files/5307422/csmLogicalReconstructMax.log)

dehann, 2020-09-30:
Solved case:
```
4.13   13:57:09.767 trafficRedirectCon   | needdownmsg P 2needdownmsg C 8needdownmsg |S| 6needdownmsg 
4.14   13:57:09.77  waitChangeOnParent   | needdownmsg P 2needdownmsg C 8needdownmsg |S| 6needdownmsg 
```

Bad case:
```
4.13   14:30:17.709 trafficRedirectCon   | needdownmsg P 2initialized C 8needdownmsg |S| 6needdownmsg 
4.14   14:30:17.714 doAllSiblingsNeedD   | needdownmsg P 2initialized C 8needdownmsg |S| 6needdownmsg 
```

dehann, 2020-09-30:
Bigger issue might be that for clique 4 both `tryInitCliq` (bad case) and `attemptDownInit` (good case) transition the status to `:initialized`.  Looking into that more

Affie, 2020-09-30:
> Ah, I tried it again with .dbg = true and that seemed to introduce enough delay

Maybe its because dbg involves io and will cause a switch to another task. Does a small sleep in the loop also fix it?

dehann, 2020-10-01:
Yes a delay in the right place will fix it, so there is a gap in current CSM.  What I previously did was use the keyword
```julia
solveTree!(...; injectDelayAfter=[4=>(13=>10);])
```

to artificially introduce delays after each step in the system and then do a success or fail criteria.  
https://github.com/JuliaRobotics/Caesar.jl/blob/master/examples/dev/devCSMHexDelayMatrix.jl

The nominal path taken can then be extracted using the Occurance.jl tools:
https://github.com/JuliaRobotics/Caesar.jl/blob/master/examples/dev/CSMOccurance.jl

Obviously, the bigger issue is that the gap in the code should not exists in the first place, so CSM needs to be refactored to get single responsibility in the code and prevent such gaps from occurring.

dehann, 2020-10-01:
okay, so here is something weird going on with clique 4.  In the "good case" (4) goes from `:needdownmsg` -> `:initialized`:
```
4.29   13:57:30.206 attemptDownInit      | needdownmsg P 2initialized C 8needdownmsg |S| 6upsolved 
4.30   13:57:31.419 rmMsgLikelihoodsAf   | initialized P 2initialized C 8needdownmsg |S| 6upsolved 
```

But during the similar situation in the "bad case" we get this:
```
4.27   14:30:31.052 attemptDownInit      | needdownmsg P 2initialized C 8needdownmsg |S| 6upsolved 
4.28   14:30:33.809 rmMsgLikelihoodsAf   | needdownmsg P 2initialized C 8needdownmsg |S| 6upsolved 
```

Now i'd like to know why in the "bad case" did initialization not occur?

dehann, 2020-10-01:
Just to note, in "bad case" from above, the status transitions to `:initialized` during a different function:
```
4.31   14:30:33.822 tryInitCliq          | needdownmsg P 2initialized C 8needdownmsg |S| 6upsolved 
4.32   14:30:33.928 slowWhileInit        | initialized P 2initialized C 8needdownmsg |S| 6upsolved 
```

dehann, 2020-10-01:
okay, more information -- so clique 3 during "good case" waits to go from `:needdownmsg --> :initialized` only after sibling clique 2 gets to `:upsolved`.  In the bad case, clique 3 changes to `:initialized` way earlier while clique 2 is still `:initialized` for quite some time.

Good case:
```
3.14   10:57:18.148 dwnInitSiblingWait   | needdownmsg P 1null        C 7needdownmsg |S| 2needdownmsg 5needdownmsg 
3.15   10:57:19.329 waitChangeOnParent   | needdownmsg P 1initialized C 7needdownmsg |S| 2needdownmsg 5needdownmsg 

3.19   10:57:49.605 dwnInitSiblingWait   | needdownmsg P 1initialized C 7needdownmsg |S| 2upsolved 5needdownmsg 
3.20   10:57:50.279 tryDwnInitCliq       | needdownmsg P 1initialized C 7needdownmsg |S| 2upsolved 5needdownmsg 
3.21   10:57:52.131 rmMsgLikelihoodsAf   | initialized P 1initialized C 7needdownmsg |S| 2upsolved 5needdownmsg 
```

Bad case:
```
3.14   10:53:11.628 dwnInitSiblingWait   | needdownmsg P 1null        C 7needdownmsg |S| 2needdownmsg 5needdownmsg 
3.15   10:53:12.641 tryDwnInitCliq       | needdownmsg P 1initialized C 7needdownmsg |S| 2needdownmsg 5needdownmsg 
3.16   10:53:15.991 rmMsgLikelihoodsAf   | initialized P 1initialized C 7needdownmsg |S| 2initialized 5needdownmsg 
```

dehann, 2020-10-01:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/d410dcc28b1f6090b91feb99a1e9a24e789d2814/src/InferDimensionUtils.jl#L380-L407

With the most troubling part the separate out:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/d410dcc28b1f6090b91feb99a1e9a24e789d2814/src/InferDimensionUtils.jl#L394

Which probably means that the priority this is on issue #910

dehann, 2020-10-04:
Okay with current master, I still get solve order as the main issue, however, the hex example can be forced to complete by adding the following simple delay:
```julia
idb = [6=>(determineCliqNeedDownMsg_StateMachine=>10);]
```

The difference in this case in the sibling init-solve order between "good" and "bad" cases.  The "good" case solves in order:
CSM:
```julia
6.25   02:15:31.771 determineCliqNeedD   | needdownmsg P 2initialized C ---- |S| 5upsolved 

Info: 01:08:12.68 | 6---6| x3 @ needdownmsg | 8j, dwnInitSiblingWaitOrder_StateMachine, 2, true, false, solord = [6, 5]
```

and the "bad" case:
```julia
6.25   01:10:30.424 determineCliqNeedD   | needdownmsg P 2initialized C ---- |S| 5initialized 

Info: 01:10:30.463 | 6---6| x3 @ needdownmsg | 8j, dwnInitSiblingWaitOrder_StateMachine, 2, true, false, solord = [5, 6]
```

The CSM sequencing for clique 6 either has sibling 5 as `:initialized` (bad) or `:upsolved` (good):
[image]

The forced delay shown above ensures that this sequence occurs (debug purpose only).  First time to compile adds enough delay to 6 as to induce the error.  Consecutive runs solve fine.  The forced delay on 6 is long enough for both precompile and already compiled cases to have CSM complete its solve.

dehann, 2020-10-04:
My current view of the problem here is that the joint synchronization between solvableDims as a channel (directly from siblings) as well as passing message/status information down from the parent is producing a deadlock condition.  Solving #910 is the best suited way for deterministic solution to remove any possibility of this deadlock occurring.

dehann, 2020-10-08:
workaround fix for current first compile test failure in IIF on current master on hex test in testBasicTreeInit.jl im locally using a delay hack while doing 910,

```julia
injectDelayBefore = [6=>(determineCliqNeedDownMsg_StateMachine=>10);]
```

See hack in code here:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/e3c0b52749c1fb75fee4e4b3b978bf01d1320bc6/test/testBasicTreeInit.jl#L53-L58

dehann, 2020-10-08:
Ah, not great this issue has been recreated with #958 

See example here:
https://github.com/JuliaRobotics/IncrementalInference.jl/pull/958#issuecomment-705665309

dehann, 2020-10-08:
We should add that test to RoME to help catch it in the future.

Affie, 2020-10-08:
Perhaps I should add this to tree init tests?
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/362e2fa808003ab6aecae4e07fc48412b5373aaf/test/testExpXstroke.jl#L27-L49

Affie, 2020-10-08:
It doesn't seem to be #958, I tested to see if was #958 or #957 and it fails before both of them.

dehann, 2020-10-12:
Obsolete with resolution of #855 and decision to only use xstroke-take for final consolidation of CSMs.

PR #936 (2020-10-01): resolve CSM cycle problem, #754
related to but does not yet fix #754
closed 2020-10-12 by dehann