JuliaRobotics/IncrementalInference.jl #755: Travis bug with ARM allow failures

Affie, 2020-06-22:
Just escalating the review to an issue so it doesn't get missed.
The bug is caused by this deletion:

@dehann Why don’t you like this?

_Originally posted by @Affie in https://github.com/JuliaRobotics/IncrementalInference.jl/pull/732_

dehann, 2020-06-23:
HI @Affie , oh, did something get deleted -- did not mean to delete that.  I was only trying to make Master and Tags be a required test to pass.  Sorry if I made a mistake there.

dehann, 2020-06-23:
Ah, wait are you talking about ARM or the Master and Tags failures.  Sounds like there are two things going on.  ARM should be allowed to fail.  MASTER AND TAGS should be brought back until its a problem again?  The DFG v0.8 switch was a big one but has previously always been MASTER AND TAGS -- I was trying to bring that part back.

dehann, 2020-06-23:
At least the ARM tests are passing at this time :-)

[image]

Affie, 2020-06-24:
Hi @dehann, because you said we don't want to modify travis every time a new minor release is made on DFG, I thought it was a permanent thing. "Master and Tags" is not the best description, in this case, it is "This branch and Tags". I added the line to not allow failures on releases (so "Tag and Tags"), but it was changed by above to not allow failures on ARM. 

We can just delete this line then:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/03d231d63fad0f7b2ea897c328a657753ca8d7c2/.travis.yml#L30

Perhaps DFG is stable enough so it will not be a problem again? we can always just modify travis again.

dehann, 2020-06-25:
> said we don't want to modify travis every time a new minor release is made on DFG
> Perhaps DFG is stable enough so it will not be a problem again? we can always just modify travis again.

Yeah did say that, although we should make sure that deprecation cycles are done properly so wanted to try bring it back and leave "Branch and Tags" in the required tests if possible.  Either way its only a few lines in Travis file so hopefully not too much of a big thing.

I have been changing these in RoME and Caesar around too, to try find best testing and longevity mix.

> in this case, it is "This branch and Tags".

That is a better name thanks!

> We can just delete this line then:

Cool that works for me!  Hopefully these files settle down a bit.  I'm currently working through 459 stuff but its slow going.

Affie, 2020-06-25:
> ...deprecation cycles are done properly...

Unfortunately, it does not help, every time a minor release is made in DFG the tests will fail because of the compare section.
I think it can still work because we do the additional tests on master and can always manually see if master passes.

PR #757 (2020-06-25): Update travis - allow failures on ARM
Close #755
commit 17eaf7e1 (2020-06-25): Close #755
closed 2020-06-25 by dehann