JuliaRobotics/IncrementalInference.jl #763: Suggestion: solveGraph in addition to (instead of) solveTree

GearsAD, 2020-07-04:
I understand that we're solving the Bayes tree, but can we add:
```
solveGraph!(dfg::G) = solveTree!(dfg)
```

dehann, 2020-07-05:
hi, i have been thinking about that too.. Does make sense.  What If we just do it as a const alias with clear docs indicating solveTree! is the workhorse.

“””
    $SIGNATURES
See solveTree!.
“””
const solveGraph! = solveTree!

The reason for emphasizing tree is ergonomic indication to the user.  The tree offers a variety of features well beyond what “solveGraph” might indicate.

GearsAD, 2020-07-06:
That works :+1: 

I was working with Jim and I he asked 'How do I solve the graph?' I wanted to reply 'solveGraph', but then had to explain what the tree is - hence the discussion.

GearsAD, 2020-07-06:
Added as you recommended and set up the PR - merge if you're good? https://github.com/JuliaRobotics/IncrementalInference.jl/pull/764

closed 2020-07-06 by dehann