JuliaRobotics/IncrementalInference.jl #783: Manifest softtype (Singleton model) for all softtypes

GearsAD, 2020-07-17:
We need to fix all softtypes to have a default constructors, and take out all state information. 

Types with issues:
```
struct ContinuousMultivariate{T1 <: Tuple} <: InferenceVariable
  dims::Int
  manifolds::T1
  ContinuousMultivariate{T}() where {T} = new()
  ContinuousMultivariate{T}(x::Int; manifolds::T=(:Euclid,)) where {T <: Tuple} = new(x, manifolds)
end
```

```
mutable struct DynPose2 <: IncrementalInference.InferenceVariable
  ut::Int64 # microsecond time
  dims::Int
  manifolds::Tuple{Symbol,Symbol,Symbol,Symbol,Symbol}
  DynPose2(;ut::Int64=-9999999999) = new(ut, 5, (:Euclid,:Euclid,:Circular,:Euclid,:Euclid))
end
```

Affie, 2020-07-17:
Just for reference: https://docs.julialang.org/en/v1/manual/types/#man-singleton-types-1

Affie, 2020-07-17:
Singleton types can work with current Pose2 for example:
```julia
struct Pose2 <: IncrementalInference.InferenceVariable
  dims::Int
  manifolds::Tuple{Symbol,Symbol,Symbol}
  Pose2() = new(3, (:Euclid, :Euclid, :Circular))
end
```
as
```julia
struct Pose2 end
getdims(::Pose2) = 3
getmanifolds(::Pose2) = (:Euclid, :Euclid, :Circular)
```

dehann, 2020-07-17:
Also see JuliaRobotics/RoME.jl#244

dehann, 2020-07-17:
Copy from Slack, @GearsAD asked if this was workable:
```julia
# TODO: Confirm that we can switch this out, instead of retrieving the complete variable.
# Original getSofttype(getVariable(dfg,lbl), solvekey)
getSofttype(v::DFGVariable) = typeof(v).parameters[1]()
```

Affie, 2020-07-17:
FIY, I don't know if the introspection is the best way in: `getSofttype(v::DFGVariable) = typeof(v).parameters[1]()`
rather use dispatch: `getSofttype(::DFGVariable{T}) where T = T`

Affie, 2020-07-17:
I wanted the same concept in the summary graphs and basically did it with a warning: 
https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/6b903b79b15c6754191cfb2b673dd89d1eb97e2c/src/services/DFGVariable.jl#L293-L296

also see: https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/237

dehann, 2020-07-17:
> rather use dispatch: getSofttype(::DFGVariable{T}) where T = T

that would be a hard type -- that's an even bigger discussion.  Going softtype allows a bit more freedom and does not require enumeration of all the factor permutations.

dehann, 2020-07-17:
see edit above:

> a related issue is how to make it easy for the user to add their own data types to the default factors. Softtype with state solved that problem, but making it a manifest or hardtype removes that option.

Affie, 2020-07-17:
This issue is closely related to the possible future use of a **'manifold type'**. I don't quite understand the whole hard-type soft-type thing yet, but that's beside the point. I can see how `ContinuousMultivariate` can maybe be referred to as a soft-type. The other types are closely related to traits of the value array (it tells the algorithm how to handle the values).

- ContinuousMultivariate: as Sam suggested, it can be solved by having users just define one for their application. ContinuousMultivariate4 or ContinuousMultivariate5, for example.

- Additional fields: I would guess the possible use of a manifold type would also require a split in the data with something like the field `ut::Int64` a bit of a grey area. This would probably require another structure specifically for this kind of data. like `metasolverdata`

So is the core of the issue: 
- How do we add user data to VariableNodeData to be used in the solver?
- Do we split out the user solver data?

some options off the top of my head:
- Use "smalldata"
   - slow
   - not available 
- use "softtype"
- use "user-meta-solver-data" field (probably quite similar to the factor meta data field?) + Singleton types (or later manifold) 
- use "inheritance"/"interface"/"traits", where you have a singleton type like Pose2 that you "inherit" and add your own data to.  
- other...

In the visualization use case, I needed the type to visualize it. There are only so many options that can currently be visualised. I only wanted Gaussian with a point, so PPE, but its useless without the softtype to tell you what is going on. With "summary graph",  `softtypename` was enough for the basic types such as Pose2 and Point2 (The singleton types). The rest is user-specific and hard (impossible) to make a generic visualiser for.

Affie, 2020-07-17:
> that would be a hard type -- that's an even bigger discussion. Going softtype allows a bit more freedom and does not require enumeration of all the factor permutations.

See edit above in the comment; to make reason a bit clearer and fix brackets.

@dehann, I don't know if I understand your comment correctly, maybe elaborate if it's still applicable after the edit?

Affie, 2020-07-17:
@GearsAD if there are not too many, perhaps this idea could work for the short term. The union is not the best and just to illustrate the idea.
```julia
getSofttype(::DFGVariable{T}) where T <:  Union{"all the singleton types"} = T()
getSofttype(::DFGVariable{T}) where T <:  Union{"all the rest"} = ...as it was before...
```

dehann, 2020-07-18:
Hi, trying to simplify and keep short issue -- see summary in first comment edit: https://github.com/JuliaRobotics/IncrementalInference.jl/issues/783#issue-659285005

dehann, 2020-07-18:
> How do we add user data to VariableNodeData to be used in the solver?
   Do we split out the user solver data?

I added my vote in first comment, and edited the Johan's suggestion above to add my support for which way forward.

GearsAD, 2020-07-19:
> Singleton types can work with current Pose2 for example:
>
> ```julia
> struct Pose2 <: IncrementalInference.InferenceVariable
>   dims::Int
>   manifolds::Tuple{Symbol,Symbol,Symbol}
>   Pose2() = new(3, [...]

Right, so this is a more explicit version of the same idea, although in this case we'd be using the singleton form of Pose2? That sounds like a reasonable way to enforce that it doesn't retain state.

GearsAD, 2020-07-19:
As in the edits in the initial description:
- I think we should provide the variables+factor in the factor evaluation for flexibility. If a factor requires variable data (or data from it's own factor) then it may be slow (is it really though?), but it may be necessary.
- I think the Singleton model from @Affie's suggestion would enforce the stateless structure, I'm just thinking we may want to serialize and save it somewhere. If we go with that, it's hardcoded so we can't do that.

dehann, 2020-07-19:
> it may be slow (is it really though?)

if we use the inheritance approach Johan suggests then we force the user to keep type safety (if I'm understanding the use right).  I was trying to do the same thing by using softtype originally, but now we need to do more things with it -- so this is a feature expansion process actually.

type stability is a major performance thing, so we do need a reasonable approach to keeping that for normal use.  An intermediate user should be able to augment their own data (like the radar images) and still have type safety.  I think if a novice is just messing around to try something, then combo of slow performance and documentation should help them advance to an intermediate user ;-)

> I'm just thinking we may want to serialize and save it somewhere

That goes back to `smalldata`.  How much should we force type stability in small data?

GearsAD, 2020-07-19:
Currently we're proposing smalldata is like Union{Int, Float64, String, Vector{Int}, Vector{Float64, Vector{String}}. I think that should be a reasonable amount of flexibility and won't hammer performance?

dehann, 2020-07-19:
Oh, that will hammer performance unfortunately, the object containing a field `::Union{...}` is the potential problem there (85% sure about that).  So either we should think of a way to harden smalldata, or as Johan suggests introduce yet another field "metadata".  I'm not too hot on the additional field thing, personally I'd rather just limit the user options on smalldata.  However, I think @Affie has something clever up his sleeve to use inheritance with "traits" to build 3rd option here that does a good enough job on both concerns (type safety and flexibility, but usage is one up from novice).

Affie, 2020-07-21:
First slightly off topic:
> Currently we're proposing smalldata is like Union{Int, Float64, String, Vector{Int}, Vector{Float64, Vector{String}}. I think that should be a reasonable amount of flexibility and won't hammer performance?

This will perform way better than a "type stable" string that has to be parsed every time.

If you limit the instability with other techniques such as adding type annotations and/or breaking up functions, it may be a good trade-off.

For example:
```julia
julia> d = Dict{Symbol, Any}()
Dict{Symbol,Any}
 with 0 entries

julia> push!(d, :a=>5)
Dict{Symbol,Any} with 1 entry:
  :a => 5

julia> push!(d, :b=>5.0)
Dict{Symbol,Any} with 2 entries:
  :a => 5
  :b => 5.0

julia> function sumD(d)
           return d[:a]::Int + d[:b]::Float64
       end
sumD (generic function with 1 method)

julia> @code_warntype sumD(d)
Variables
  #self#::Core.Compiler.Const(sumD, false)
  d::Dict{Symbol,Any}

Body::Float64
1 ─ %1 = Base.getindex(d, :a)::Any
│   %2 = Core.typeassert(%1, Main.Int)::Int64
│   %3 = Base.getindex(d, :b)::Any
│   %4 = Core.typeassert(%3, Main.Float64)::Float64
│   %5 = (%2 + %4)::Float64
└──      return %5
```

Affie, 2020-07-21:
> Right, so this is a more explicit version of the same idea, although in this case we'd be using the singleton form of Pose2? That sounds like a reasonable way to enforce that it doesn't retain state.

The reason to rather use a singleton is it can only have one instance. eg 
```julia
julia> Pose2() === Pose2()
true
```

> I think the Singleton model from @Affie's suggestion would enforce the stateless structure, I'm just thinking we may want to serialize and save it somewhere. If we go with that, it's hardcoded so we can't do that.

Currently, `Pose2` requires RoME to use, so I don't think it would make a difference. It makes it easier as you can just serialize the name, ie "Pose2" and not any fields.  So `getSofttype(::DFGVariable{T}) where T = T()` will still work.

Affie, 2020-07-21:
Breadcrumb for function name: `issingletontype`

Affie, 2020-07-21:
@dehann, is this the way it's added to factors:

https://github.com/JuliaRobotics/IncrementalInference.jl/blob/e9351e7140c3fcd44676cb6f65637f89f06f72ac/src/FactorGraphTypes.jl#L147-L160

And related to performance #383 and the "Vector of abstract".

Affie, 2020-07-21:
Perhaps we can split it completely as a new abstract type? (But you know me, I like abstracting everything)
```julia
struct  MyTimeVaribleUserData <: AbstractVariableUserSolverData
   utime::Int
end

#and have

variableuserdata::Vector{T} where T<: AbstractVariableUserSolverData
```
If the options are a Dictionary (like small data) or a Struct, my vote is for struck since this will be called a lot and we want it in the stack. 

PS. I'm just writing a bunch of options as a brainstorming exercise when I think of them.

Affie, 2020-07-22:
@dehann, why is `utime` not a measurement in the factor?

Affie, 2020-07-23:
@GearsAD, I think I missed it, what was your original reason to want softtype stateless?

GearsAD, 2020-07-28:
So I think in an ideal world we should remove `FactorMetadata` and head in this direction:

```
function (vp2vp2::VelPoint2VelPoint2{D})(
                res::Array{Float64},
                variables::Vector{DFGVariable},
                idx::Int,
                meas::Tuple,
                Xi::Array{Float64,2},
                Xj::Array{Float64,2}  ) where D
  #
  z = meas[1][:,idx]
  xi, xj = Xi[:,idx], Xj[:,idx]
  # change in time from microseconds with DynPoint2(ut=1_000_000) to seconds
  dt = (variables[2].timestamp - variables[1].timestamp)*1e-6   # roughly the intended use of userdata
...
```

dehann, 2020-07-28:
```julia

struct FactorMetadata_NEW{USER}
  factoruserdata
  cacheData::Vector{USER}
  fullVariable::Vector{DFGVariable}
  solvefor::Symbol
  variablelist::Vector{Symbol}
  dbg::Bool
end

# so naive replacement for timestamp is to use

function (::AwesomeFactor)(..., fmd::FactorMetadata, ...)

t1 = getTimestamp(fmd.fullVariable[1])
t2 = getTimestamp(fmd.fullVariable[2])

end

# so intermediate user replacement for timestamp is to use

struct MyDynPoseCache
  timestamp::Int64
end

# user must do this, and IIF must somehow know to call this... TBD
prepCache(::Type{MyDynPoseCache}, ::DfgVariable, timestamp) = getTimestamp(fullVariable)

function (::AwesomeFactor)(..., fmd::FactorMetadata, ...)
  t1 = fmd.cacheData[1].timestamp
  t2 = fmd.cacheData[2].timestamp

  # do matj
end

## What the hell, let me do it myself...

struct MyDynPose2Factor
  .. do it all inthe factor
end

```

dehann, 2020-07-28:
the `prepCache` function (we still need a better name for it) will likely be called around here:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/a43adec70e837e076eb98ea078fd633b991a7dd5/src/ApproxConv.jl#L75

Affie, 2020-07-28:
Is it ok to have continuous scalar only represent Euclid, ie `ContinuousScalar() = new(1, (:Euclid,))`

```julia
struct ContinuousScalar <: InferenceVariable
  dims::Int
  manifolds::Tuple{Symbol}
  ContinuousScalar(;manifolds::Tuple{Symbol}=(:Euclid,)) = new(1, manifolds)
end
```

Since we have Sphere1 for (:Circular,)

```julia
struct Sphere1 <: InferenceVariable
  dims::Int
  manifolds::Tuple{Symbol}
  Sphere1() = new(1, (:Circular,))
end
```

We can also do ContinuousMultivariate as all Euclidian

```julia
julia> struct ContinuousMultivariateEuclid{N} <: InferenceVariable end
julia> function Base.getproperty(::ContinuousMultivariateEuclid{N}, f::Symbol) where N
         if f == :dims
           return N
         elseif f == :manifolds
           return ntuple(i -> :euclid, 5) 
         end
       end

julia> a = ContinuousMultivariateEuclid{4}()
julia> a.dims
4
julia> a.manifolds
(:euclid, :euclid, :euclid, :euclid, :euclid)

```

dehann, 2020-07-28:
Hi, yes I think thats fine thanks -- we can later add things like `ContinuousCircular`?

dehann, 2020-07-28:
i like the `::ContinuousMultivariateEuclid{N}` since that is similar to `::NTuple{N}` and `::Array{N}`

dehann, 2020-07-28:
Oh, just use `:Euclid`.  This will all become part of the JuliaRobotics/RoME.jl#244 Manifold.jl consolidation story.

Affie, 2020-07-30:
I wrote down a trait-based example we can look at using the softtypes for other things such as visualization. 
```julia
abstract type VisualizerTrait end
struct SE{N} <: VisualizerTrait end
struct Cartesian{N} <: VisualizerTrait end
VisualizerTrait(::T) where T = error("Type $T doesn't implement the VisualizerTrait")

VisualizerTrait(::Pose2) = SE{2}()
VisualizerTrait(::Pose3) = SE{3}()
VisualizerTrait(::Point2) = Cartesian{2}()
VisualizerTrait(::Point3) = Cartesian{3}()

visualize(x) = visualize(VisualizerTrait(x), x)
visualize(::SE{N}, x) where N = @info "Visualizing SE$N..."
visualize(::Cartesian{N}, x) where N = @info "Visualizing Cartesian $N..."

Xs = [Pose2(), Pose3(), Point2(), Point3()]

julia> visualize.(Xs)
[ Info: Visualizing SE2...
[ Info: Visualizing SE3...
[ Info: Visualizing Cartesian 2...
[ Info: Visualizing Cartesian 3...

```

dehann, 2020-07-31:
i like that!

Affie, 2020-08-03:
Added advantage, micro performance improvement. It might be negligible but this way the constant propagation can be used:

```julia
#the 2 versions of:
@code_lowered getDims(p3)
# old
CodeInfo(
1 ─ %1 = Base.getproperty(p, :dims)
└──      return %1
)
# new
CodeInfo(
1 ─     return 6
)
```

dehann, 2020-08-03:
nice!

dehann, 2020-08-03:
Decision for #784 made, consolidating work effort in this issue.

Affie, 2020-08-07:
Partially addressed by #823 - with full variables.
Deprecation stared to change softtypes to singletons

dehann, 2020-08-12:
lol https://github.com/JuliaRobotics/RoME.jl/issues/62

dehann, 2020-08-14:
see JuliaRobotics/DistributedFactorGraphs.jl#603

Affie, 2020-08-17:
Stared the deprecation process in #841.

dehann, 2020-10-23:
Too big a change to also include in v0.17.0, punting down one to v0.18.0 since this will require a breaking change upstream from DFG v0.11.0 which is not available yet.

dehann, 2020-12-09:
xref #825

closed 2020-12-09 by dehann