JuliaRobotics/IncrementalInference.jl #784: Discussion: Pass variables to factor evaluation so that we can use variable data?

GearsAD, 2020-07-17:
I'd like to use the data from variables in the factor evaluation, e.g. Radar data. I think we should pass down the variables into the factor evaluation, so that I could potentially use smallData or bigData.

dehann, 2020-07-17:
Remember this part too about user desire for more "metadata" in a variable: https://github.com/JuliaRobotics/IncrementalInference.jl/issues/783#issuecomment-660234861

dehann, 2020-07-19:
moving to v0.15.0 as more realistic.  Hopefully can make 783 decision by v0.14.0

dehann, 2020-08-03:
resolved with discussion around and following https://github.com/JuliaRobotics/IncrementalInference.jl/issues/783#issuecomment-665080114

closed 2020-08-03 by dehann

dehann, 2021-02-06:
Punchline is to use the new `CalcFactor` approach, and hopefully stable API defined in v0.21.  See #467 as relatively well linked issue discussion on what the API should be.  All metadata (variables, smalldata, entry=>data, etc.) is available through the `CalcFactor` approach.  This is both during residual factor evaluation and `getSample`.