JuliaRobotics/IncrementalInference.jl #812: Travis Conditional allow_failure MASTERS on release/v0.x

dehann, 2020-07-27:
Rather than hard yes or no on DEV MASTERS when running Travis for release/v0.x branches, perhaps there is a way to do a conditional allow_failure setup where stable releases are still tested against upsteam masters, but a failure is allowed.

## Suggestions / Ideas

- One idea is to have to `- jobs` which are nearly identical for DEV MASTERS case, the one includes an allowable failure flag, the other doesn't.  Only one of them runs in the case of `release/v0.x` and the usual (does not allow failures) in all other cases.

dehann, 2020-07-27:
FYI JuliaRegistries/Tagbot#163

Affie, 2020-07-30:
I'm not completely following the issue.

I had similar conditional failures before and I added it as a comment for future reference (so I don't have to figure it out again):
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/91d97bee06c0bffa6df971bde5776878a1ac56b5/.travis.yml#L30-L32
The problem with this was it's only needed if the minor version changes while in the 0.x versions. So ideally it should look at the tags also. Perhaps we can find a way to bring in patch/minor. It seems to be a common problem for a lot of the larger split ecosystems, such as Images.jl

Better names for what we have could be `This and Masters` and `This and Tags`
Currently:
- `This and Masters` 
    - Not tested on release branches
- `This and Tags` 
    - No failures allowed making the test fail if there is a minor release but can be manually checked on `This and Masters`

So perhaps the solution is "One Shall Pass"

Are you suggesting:
- `This and Masters` 
    - Allow failures on release branches
- `This and Tags` 

I'm against running so many tests if it's not needed (takes too long).