JuliaRobotics/IncrementalInference.jl #838: auto promote ::DateTime to ::ZonedDateTime

dehann, 2020-08-15:
first suggested by folks elsewhere, but to formalize:  we should permanently do automation inside IIF.addVariable! IIF.addFactor! to promote `::DateTime` to `::ZonedDateTime(dt, localzone())`, since the DFG standard requires ISO 8601 ZonedDateTime for all time stamping.

Affie, 2020-08-15:
There are separate constructors that takes a datetime (and keyword timezone) and converts is already.

Affie, 2020-08-15:
There were promotion rules the other way around (zoneddatetime to datetime) to help with depreciation. But that is type piracy and was only temporary.

Affie, 2020-08-15:
Just for reference: 
https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/ff27aa4b8550f7ac3990aaeb0436062767378b82/src/entities/DFGVariable.jl#L269-L273

https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/ff27aa4b8550f7ac3990aaeb0436062767378b82/src/services/CommonAccessors.jl#L73

dehann, 2020-08-15:
cool thanks! we should probably make the DFGVariable function use dispatch rather than an if statement, not important to do now though.

closed 2020-08-15 by dehann

Affie, 2020-08-15:
Jip, good idea. Didn't you want to add it to the `addVariable!` `addFactor!` functions in IIF?