JuliaRobotics/IncrementalInference.jl #843: Tree init gets freezes

Affie, 2020-08-18:
```julia
fg = generateCanonicalFG_lineStep(4)
getSolverParams(fg).graphinit = false
getSolverParams(fg).treeinit = true
tree, smt, hist = solveTree!(fg)
```

Affie, 2020-08-18:
result
```julia
[ Info: Solving over the Bayes (Junction) tree.

[ Info: Building Bayes net...
[ Info: 
[ Info: Eliminating x4
[ Info: ===============
[ Info: 
[ Info:  - Removing x4
[ Info: 
[ Info: Eliminating lm0
[ Info: ===============
[ Info: 
[ Info:  - Removing lm0
[ Info: 
[ Info: Eliminating lm4
[ Info: ===============
[ Info: 
[ Info:  - Removing lm4
[ Info: Looking at x2lm4f2
[ Info:    - Breaking link x2lm4f2->lm4...
[ Info:      - Original links: [:x2, :lm4]
[ Info:      - New links: [:x2]
[ Info: x2lm4f2 still has links to other variables, readding it back...
[ Info: 
[ Info: Eliminating x0
[ Info: ===============
[ Info: 
[ Info:  - Removing x0
[ Info: Looking at x0x2f2
[ Info:    - Breaking link x0x2f2->x0...
[ Info:      - Original links: [:x0, :x2]
[ Info:      - New links: [:x2]
[ Info: x0x2f2 still has links to other variables, readding it back...
[ Info: 
[ Info: Eliminating x2
[ Info: ===============
[ Info: 
[ Info:  - Removing x2
[ Info: Looking at x2f2
[ Info:    - Breaking link x2f2->x2...
[ Info:      - Original links: [:x2]
[ Info:      - New links: Symbol[]
[ Info: x2f2 doesn't have any other links, not adding it back...
[ Info: Looking at x2f1
[ Info:    - Breaking link x2f1->x2...
[ Info:      - Original links: [:x2]
[ Info:      - New links: Symbol[]
[ Info: x2f1 doesn't have any other links, not adding it back...
[ Info: Looking at x2f3
[ Info:    - Breaking link x2f3->x2...
[ Info:      - Original links: [:x2]
[ Info:      - New links: Symbol[]
[ Info: x2f3 doesn't have any other links, not adding it back...
[ Info: Looking at x2f4
[ Info:    - Breaking link x2f4->x2...
[ Info:      - Original links: [:x2]
[ Info:      - New links: Symbol[]
[ Info: x2f4 doesn't have any other links, not adding it back...
[ Info: Staring the Bayes tree construction from Bayes net
[ Info: Adding x2 to tree...
[ Info: Adding x0 to tree...
[ Info: Adding lm4 to tree...
[ Info: Adding lm0 to tree...
[ Info: Adding x4 to tree...
Find potential functions for each clique
[ Info: Get potentials 2| lm4,x4,: x2,
[ Info: using all factors connected to frontals and attached to separator
[ Info: finding all frontals for down WIP
[... 1107 lines cut ...]
└ @ DistributedFactorGraphs ~/.julia/packages/DistributedFactorGraphs/rH2PI/src/services/AbstractDFG.jl:955
┌ Warning: Tuple{KDE,Floa64} specific version of addMsgFactors! is deprecated, use LikelihoodMessage version instead.
└ @ IncrementalInference ~/.julia/packages/IncrementalInference/FwCTW/src/Deprecated.jl:34
┌ Warning: Factor x0x2f1 will be an orphan in the destination graph, and therefore not added.
└ @ DistributedFactorGraphs ~/.julia/packages/DistributedFactorGraphs/rH2PI/src/services/AbstractDFG.jl:955
┌ Warning: Factor x2lm4f1 will be an orphan in the destination graph, and therefore not added.
└ @ DistributedFactorGraphs ~/.julia/packages/DistributedFactorGraphs/rH2PI/src/services/AbstractDFG.jl:955
┌ Warning: Factor x2x4f1 will be an orphan in the destination graph, and therefore not added.
└ @ DistributedFactorGraphs ~/.julia/packages/DistributedFactorGraphs/rH2PI/src/services/AbstractDFG.jl:955
┌ Warning: Factor x2lm0f1 will be an orphan in the destination graph, and therefore not added.
└ @ DistributedFactorGraphs ~/.julia/packages/DistributedFactorGraphs/rH2PI/src/services/AbstractDFG.jl:955
┌ Warning: Tuple{KDE,Floa64} specific version of addMsgFactors! is deprecated, use LikelihoodMessage version instead.
└ @ IncrementalInference ~/.julia/packages/IncrementalInference/FwCTW/src/Deprecated.jl:34
[ Info: cycleInitByVarOrder! -- varorder=[:lm4, :x4, :x2]
[ Info: var.label=lm4 is initialized=false
[ Info: try doautoinit! of lm4
[ Info: init with useinitfct Symbol[]
[ Info: var.label=x4 is initialized=false
[ Info: try doautoinit! of x4
[ Info: init with useinitfct Symbol[]
[ Info: var.label=x2 is initialized=false
[ Info: try doautoinit! of x2
[ Info: init with useinitfct Symbol[]
[ Info: cycleInitByVarOrder!, retval=false
```

dehann, 2020-08-19:
cool thanks for the example.  Tree init is still under development, but it’s pretty close to ready so shouldn’t be too far to go.  Problem likely due to prioritization between siblings which doesn’t work right if there are no siblings below the root.

dehann, 2020-09-19:
After #908 the CSM sequencing was changed so not worth debugging this problem.  Targeting v0.16 to resolve this issue.

closed 2020-09-19 by dehann

dehann, 2020-09-19:
[image]

dehann, 2020-09-19:
might be related to #754

Affie, 2020-09-22:
It works after #459