JuliaRobotics/IncrementalInference.jl #855: fetch vs take! on message channels

Affie, 2020-08-24:
Currently parametric works with a take! and save (for later use and debugging) model. (or used to, it's broken)

Affie, 2020-08-25:
The "take! and save" approach in parametric is based on:
- The "4-stroke" state machine with messages going all the way for initialization also (leaves <--> root)
    - Based on asynchronous vs synchronous message passing algorithms.
- Therefore, cliques only calculate messages once per stroke.
- I wanted to rely on the **channels** alone for **synchronisation** and wait for all children to send a message.

dehann, 2020-08-26:
Now that i'm thinking about 4-stroke more, I can think of one case that might be tricky to do without "cascading", but lets worry about that after consolidation.

dehann, 2020-09-25:
> synchronisation I also mean to wait on

For tree init the status needs to be checked multiple times, and later messages need to be fetched more than once -- this make the take! on one channel only harder i think.  I was not able to resolve tree init without switching to a fetch model with Condition to allow a wait cycle again.

>  channels are basically just wrappers for convenience around conditions (with a buffer)

Yes and no, `Conditions` are edge triggered while `Channels` are level triggered.  It's discussed somewhere in the Julia docs.

You'll see in CSM there are states that are `slowXYZ`.  This is usually a wait on condition with checks for some kind of status, and also short loops in the CSM should a certain criteria be met.  This is sometimes required for treeinit where a hard block until upsolved is not always possible, or when siblings need to decide on a down init order (thats in the cascading model).

Affie, 2020-09-29:
> Yes and no, Conditions are edge triggered while Channels are level triggered

Yes, you are correct. For me, that is an advantage, with level triggering cliques don't need to be waiting when the message is sent.

dehann, 2020-09-30:
Just confirming, for the take model you send a copy of the down message to each child?

Affie, 2020-09-30:
Yes. As you know there is a channel per edge. Currently for down the same message is sent on all down channels. I considered a possible future need to send a specific message to only one child, but that didn’t seem necessary yet.

Affie, 2020-09-30:
The cliques are scheduled by the repeating pattern:
- ....
- send up
- wait for down
- do some down stuff
- send down
- wait for up
- do some up stuff
- ....

Affie, 2020-10-07:
WIP take! storyboard
[image]

# take! stating UP sequence 
- Do all prep (MarginalizedRecycle, IncrementalRecycle, BuildSubgraph, etc)
- Wait for up message if needed (leaves start)
- Try up init
    - If priors try init
    - Elseif linear on manifold (TODO, calculate for differential only)
    - Else can't init
- Send message up
- Wait for down message if needed (root start)
- Branch for down
    - If all children up-solved down solve
    - else try down init
- Try down init
- Send message down
- Wait for up message if needed (leaves start)

... If cliques can now be initialized with new messages upsolves starts, otherwise the init sequence is continued until all are initialized

- Do upsolve
- Send message up
- Wait for down message if needed (root start)
- Downsolve
- Send message down
- Update from subgraph and exit CSM

dehann, 2020-10-15:
Ah great thank you, I added a link from IIF Wiki CSM-Storyboard above.

dehann, 2020-10-15:
Will be interesting to see this expand if cascaded down init returns as shortcut over x-stroke (if that were to happen again).

dehann, 2020-10-18:
think we are +2 for take!-only model,

@Affie can confirm

closed 2020-10-18 by dehann