JuliaRobotics/IncrementalInference.jl #882: problem with loading the example "RobotFourDoor.jl"

EstherVorkin, 2020-09-07:
I'm trying to load this example but have the following issues:
- ERROR: UndefVarError: inferOverTree! not defined
- ERROR: UndefVarError: Odo not defined
- ERROR: UndefVarError: Obsv2 not defined

What can I do? I've updated the repository today.

Thanks in advance

Affie, 2020-09-07:
Thanks for opening the issue.
The example is a bit outdated with the functions you mention deprecated:

Odo -> LinearConditional
Obsv2 -> Prior
inferOverTree! -> solveTree!

I'll see if it's a quick fix to update the example.

dehann, 2020-09-09:
Hi @EstherVorkin , thanks for posting.  In the mean time, please see the e.g. `multihypo=[1;0.5;0.2;0.3]` method too:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/master/test/testMultiHypo2Door.jl

Thanks @Affie for taking a look and updating the quite outdated example!

closed 2020-09-10 by dehann

EstherVorkin, 2020-09-12:
Good day,
Thanks for your reply!
Is the test should return error while running?
I tried both tests: testMultiHypo2Door and foorDoorTest

EstherVorkin, 2020-09-12:
I always get the following error: ERROR: UndefVarError: exists not defined
Maybe it's a problem with my code. What could it be ?

Best regards.

PR #887 (2020-09-10): Fix RobotFourDoor example
close #882

~~needs https://github.com/JuliaRobotics/RoMEPlotting.jl/pull/122~~