JuliaRobotics/IncrementalInference.jl #895: ERROR: UndefVarError: exists not defined

EstherVorkin, 2020-09-13:
This error appears in the code each time I try to run one of the tests:
basicGraphOperations.jl - line 13 and 15
fourDoorTest.jl
testMultiHypo2Door.jl

Can you help with this one?

dehann, 2020-09-14:
Hi, that seems quite strange, `exists` should be there:
```julia
julia> using IncrementalInference
julia> exists
exists (generic function with 6 methods)

help?> exists
search: exists exitStateMachine extractdistribution DiscreteMatrixDistribution implements_vertex_list

  exists(dfg, node)
  

  True if the variable or factor exists in the graph.
```

Which version of the code are you one?  And perhaps a copy of the error and error stack that you getting perhaps please? 

Also note a recent issue #893 (a regression which was accidentally introduced into the latest versions) -- we working to get that fixed.

EstherVorkin, 2020-09-14:
I have the updated version of Julia 15.1 and downloaded the latest version of the code from here.
When I'm trying to do as you did:

julia> using IncrementalInference
[ Info: FunctionalStateMachine.jl is adding Graphs.jl related tools (Visualization).

julia> exists
ERROR: UndefVarError: exists not defined

Affie, 2020-09-14:
Please post the results of the following:

```julia
julia> versioninfo()
```
and

```julia
julia> ]status
```

EstherVorkin, 2020-09-14:
version info:
Julia Version 1.5.1
Commit 697e782ab8 (2020-08-25 20:08 UTC)
Platform Info:
  OS: Windows (x86_64-w64-mingw32)      
  CPU: Intel(R) Core(TM) i5-6200U CPU @ 2.30GHz
  WORD_SIZE: 64
  LIBM: libopenlibm
  LLVM: libLLVM-9.0.1 (ORCJIT, skylake)
Environment:
  JULIA_EDITOR = "C:\Users\EstherVorkin\AppData\Local\atom\app-1.51.0\atom.exe"  -a
  JULIA_NUM_THREADS = 2

]status:
Status `C:\Users\EstherVorkin\.julia\environments\v1.5\Project.toml`
  [c52e3926] Atom v0.12.21
  [159f3aea] Cairo v1.0.5
  [186bb1d3] Fontconfig v0.4.0
  [c91e804a] Gadfly v1.3.0
  [7073ff75] IJulia v1.21.3
  [904591bb] IncrementalInference v0.10.2
  [e5e0dc1b] Juno v0.8.3
  [2472808a] KernelDensityEstimate v0.5.2
  [295af30f] Revise v2.7.5
  [91fb55c2] RoME v0.6.1
  [238d586b] RoMEPlotting v0.2.2
  [8dfed614] Test

Affie, 2020-09-14:
Thanks, that helps. 
I see you are using windows, the code is not tested on windows, I dought everything will work, but haven't tried it myself.
The easiest will be to run it in Linux. 

The package versions are quite out of date and have to be updated to run the examples.
The current versions are: 
IncrementalInference v0.15.3
RoME v0.9.0

EstherVorkin, 2020-09-14:
I will try to run the updated versions and on linux. Will update you today
if it helps.

Thanks for now

EstherVorkin, 2020-09-15:
I have installed VM with linux on it and downoaded the latest versions of all.
The latest IncrementalInference version is  v0.11.2 and RoME version is v0.6.4

The same error appears now

Affie, 2020-09-15:
It seems you found an issue with the compatibility of the latest packages. It looks like the package manager is struggling to resolve some of the dependencies.

If I add in a new environment, IIF is downgraded `↓ IncrementalInference v0.15.4 ⇒ v0.11.2`:
```julia
(tmpenv) pkg> add IncrementalInference
  Resolving package versions...
Updating `~/.julia/dev/IncrementalInference/~/Documents/juliasandbox/tmpenv/Project.toml`
  [904591bb] + IncrementalInference v0.15.4

(tmpenv) pkg> add RoMEPlotting
  Resolving package versions...
  Installed RoMEPlotting ─ v0.2.2
  Installed RoME ───────── v0.6.4
Updating `~/.julia/dev/IncrementalInference/~/Documents/juliasandbox/tmpenv/Project.toml`
  [904591bb] ↓ IncrementalInference v0.15.4 ⇒ v0.11.2
  [238d586b] + RoMEPlotting v0.2.2
Updating `~/.julia/dev/IncrementalInference/~/Documents/juliasandbox/tmpenv/Manifest.toml`
  [621f4979] + AbstractFFTs v0.5.0
  [13072b0f] + AxisAlgorithms v1.0.0
  [6e34b625] + Bzip2_jll v1.0.6+4
  [159f3aea] + Cairo v1.0.5
  [83423d85] + Cairo_jll v1.16.0+5
  [49dc2e85] + Calculus v0.4.1
  [34da2185] ↓ Compat v3.16.0 ⇒ v2.2.1
  [a81c6b42] + Compose v0.8.2
  [d38c429a] + Contour v0.5.5
  [7ad07ef1] + CoupledFields v0.2.0
  [01453d9d] + DiffEqDiffTools v0.13.0
  [b5cc3c7e] ↓ DistributedFactorGraphs v0.10.2 ⇒ v0.7.7
  [31c24e10] ↑ Distributions v0.23.8 ⇒ v0.23.11
  [2e619515] + Expat_jll v2.2.7+5
  [e2ba6199] - ExprTools v0.1.2
  [8f5d6c58] - EzXML v1.1.0
  [7a1cc6ca] + FFTW v1.2.4
  [f5851436] + FFTW_jll v3.3.9+5
  [1a297f60] ↑ FillArrays v0.8.14 ⇒ v0.9.6
  [186bb1d3] + Fontconfig v0.4.0
  [a3f928ae] + Fontconfig_jll v2.13.1+13
  [d7e528f0] + FreeType2_jll v2.10.1+4
  [559328eb] + FriBidi_jll v1.0.5+5
  [c91e804a] + Gadfly v1.3.0
  [78b55507] + Gettext_jll v0.20.1+6
  [7746bdde] + Glib_jll v2.59.0+3
  [a2bd30eb] + Graphics v1.0.2
  [3b182d85] + Graphite2_jll v1.3.13+3
  [2e76f6c2] + HarfBuzz_jll v2.6.1+9
  [a1b4810d] + Hexagons v0.2.0
  [a51ab1cf] + ICU_jll v67.1.0+1
  [904591bb] ↓ IncrementalInference v0.15.4 ⇒ v0.11.2
  [9b13fd28] + IndirectArrays v0.5.1
  [1d5cc7b8] + IntelOpenMP_jll v2018.0.3+0
  [a98d9a8b] + Interpolations v0.12.10
  [c8e1da08] + IterTools v1.3.0
  [e5e0dc1b] + Juno v0.8.3
  [5ab0869b] + KernelDensity v0.5.1
  [c43967c8] + KernelDensityEstimatePlotting v0.1.7
  [dd4b983a] + LZO_jll v2.10.0+2
  [e9f186c6] + Libffi_jll v3.2.1+3
  [d4300ac3] + Libgcrypt_jll v1.8.5+3
  [7add5ba3] + Libgpg_error_jll v1.36.0+2
  [4b2f31a3] + Libmount_jll v2.34.0+2
  [38a345b3] + Libuuid_jll v2.34.0+6
  [4345ca2d] + Loess v0.5.0
  [856f044c] + MKL_jll v2020.2.254+0
  [442fdcdd] + Measures v0.3.1
  [e89f7d12] + Media v0.5.0
  [78c3b35d] - Mocking v0.7.1
  [6fe1bfb0] + OffsetArrays v1.2.0
  [429524aa] ↓ Optim v1.2.0 ⇒ v0.18.1
  [2f80f16e] + PCRE_jll v8.42.0+3
  [36c8627f] + Pango_jll v1.42.4+9
  [d96e819e] ↓ Parameters v0.12.1 ⇒ v0.10.3
  [30392449] + Pixman_jll v0.38.4+4
  [c84ed2f1] + Ratios v0.4.0
  [3cdcf5f2] - RecipesBase v1.1.0
  [91fb55c2] + RoME v0.6.4
  [238d586b] + RoMEPlotting v0.2.2
  [6038ab10] + Rotations v1.0.1
  [992d4aef] + Showoff v0.3.1
  [2913bbd2] ↓ StatsBase v0.33.1 ⇒ v0.32.2
  [f269a46b] - TimeZones v1.3.2
  [3a884ed6] - UnPack v1.0.2
  [efce3f68] + WoodburyMatrices v0.5.2
  [aed1982a] + XSLT_jll v1.1.33+3
  [4f6342f7] + Xorg_libX11_jll v1.6.9+3
  [0c0b7dd1] + Xorg_libXau_jll v1.0.9+3
  [a3789734] + Xorg_libXdmcp_jll v1.1.3+3
  [1082639a] + Xorg_libXext_jll v1.3.4+3
  [ea2f1a96] + Xorg_libXrender_jll v0.9.10+3
  [14d82f49] + Xorg_libpthread_stubs_jll v0.1.0+2
  [c7cfdc94] + Xorg_libxcb_jll v1.13.0+2
  [c5fb5394] + Xorg_xtrans_jll v1.4.0+2
  [b53b4c65] + libpng_jll v1.6.37+5
  [9abbd945] + Profile
```
I'll try and track it down.

Affie, 2020-09-15:
It looks to be related to an issue I had with Gadfly compatibility. 

for now you can try adding the masters of Gadfly and RoME
```julia
(tmpenv) pkg> add Gadfly#master
(tmpenv) pkg> add RoME#master
(tmpenv) pkg> add IncrementalInference
(tmpenv) pkg> add RoMEPlotting
```
```

EstherVorkin, 2020-09-15:
Ok thank you. Added the masters. Will wair for other updates.
Thanks for all the help

EstherVorkin, 2020-09-15:
Thanks for all your help
I've made the last changes and now I have the needed versions.
Exists still now defined though

Affie, 2020-09-15:
That is really strange. `exists` is definitely exported. 

Is it found if you say `using DistributedFactorGraphs`

EstherVorkin, 2020-09-15:
The "basicGraphOperation.jl" test works now ! The test passed twice.
Now the only problem detected that it doesn't recognise "FactorGraph" in "fourDoorTest.jl"

Affie, 2020-09-15:
Perhaps try with a new environment and a new julia instance and make sure the packages are at the correct versions with `]status`
FactorGraph was removed a while back.

dehann, 2020-09-18:
Hi please note that any changes for v0.15.x will now have to be done as a backport since `master` has moved on to and targeting v0.16.0.

Affie, 2020-10-17:
A new Gadfly has since been tagged and IIF, RoME, RoMEPlotting are all updated.

closed 2020-10-17 by Affie